node-opcua/node-opcua #223: Create a node-opcua client which invokes an OPC-UA method in a server

juancadavid, 2016-06-28:
Hello,

I'm looking to develop a Node-opcua client which invokes a OPC-UA method in a server. I have not found an example on how to do this. I've found how to create a server a server with a method, but not a client with a method call.
Could you please refer me to an example?

Thanks in advance,

Juan

KadriUmay, 2016-07-05:
Hi,

I've tried the sample "creating a simple server with method" and was able to see it in UA Expert client. When I try to write the parameters back it says Bad Not Writeable. Do I need to add any parameters to the input parameters definition to elevate the access level?

Thanks

erossignon, 2016-08-30:
@KadriUmay  can you be more specific about what you do in "when I to write the parameters back".
Which parameter ? how ?

closed 2019-03-26 by erossignon