node-opcua/node-opcua #224: Problems with Simple Server on Intel Edison

Alexi16, 2016-06-29:
Hello!

I followed the steps in the "Creating a Simple Server" tutorial to get an OPC UA server on my Intel Edison. I am able to start the server. But I can't connect it to my OPC UA client (Sample Client from OPC Foundation, running on Windows 7).

With Wireshark I found out that the GetEndpointRequest works. The problem is at creating the communication channel. The last message sent was ActivateSessionRequest. Afterwards, I get this error on the Intel Edison:

```
/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/datamodel/variant_tools.js:90
            return typeof value === "string" || util.isNullOrUndefined(value);
                                                     ^
TypeError: Object #<Object> has no method 'isNullOrUndefined'
    at isValidScalarVariant (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/datamodel/variant_tools.js:90:54)
    at isValidVariant (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/datamodel/variant_tools.js:151:20)
    at Object.Variant_Schema.construct_hook (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/schemas/Variant_schema.js:384:18)
    at new Variant (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/_generated_/_auto_generated_Variant.js:99:22)
    at UAVariable._bindComponent.prop.bindVariable.timestamped_get [as _timestamped_get_func] (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/address_space/ua_variable.js:1350:41)
    at UAVariable._Variable_bind_with_timestamped_get (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/address_space/ua_variable.js:915:33)
    at bind_getter (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/address_space/ua_variable.js:1034:45)
    at UAVariable.bindVariable (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/address_space/ua_variable.js:1210:5)
    at _bindComponent (/home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/address_space/ua_variable.js:1323:10)
    at /home/myedisonserver/nodeexamples2/myserver/node_modules/node-opcua/lib/address_space/ua_variable.js:1461:17
```

What does this error mean? Is it possible that nodeopcua needs a special version of node? My version on the Intel Edison is 0.10.28 for node and 1.4.9 for npm.

erossignon, 2016-06-29:
Good spot. the issue is with nodejs 0.10.28 that doesn't provide the isNullOrUndefined utility function. 

Note that we would like to drop support for very old version of node. Do you know what is the newest version of node currently supported by Intel Edison ?

tchekoto, 2016-06-30:
Hello,

Have you tried "opkg update" ?

I do not have Intel edison any more, but if I remember well, it was node
0.10.38.

William.

Alexi16, 2016-07-02:
Hello,

thank you for your help!

I tried "opkg update", but it didn't update the node version (it's still v0.10.28).

I understand that you want to drop support for old node versions. Unfortunately, I don't know how to update the version. In a forum I read that it's not possible to update higher than 0.12.x due to the Yocto Image. 

But what helped me was the change you porposed in the variant_tools.js file. It's working now, thank you very much!

closed 2016-07-03 by erossignon