node-opcua/node-opcua #225: Client cannot connect to the demo server

lhf552004, 2016-06-30:
---
OS: Windows7 professional X64;
node version: v4.4.7
openSSL: Win64 OpenSSL v1.0.2h Light, is from https://slproweb.com/products/Win32OpenSSL.html 
---
There is an error when I try to connect to server from simple client.

$ node bin/simple_client.js -e "opc.tcp://WIN-74C3L8J7J4L:26543" -n="ns=1;s=EVR2.system.RTC_SEC"

securityMode        =  NONE
securityPolicy      =  None
timeout             =  20000
 monitoring node id =  ns=1;s=PumpSpeed
 connecting to  opc.tcp:\WIN-74C3L8J7J4L:26543
C:\Projects\node-opcua\lib\nodeopcua.js:98
        throw new Error("Invalid endpoint url ",endpoint_url);
        ^

Error: Invalid endpoint url
    at parseEndpointUrl (C:\Projects\node-opcua\lib\nodeopcua.js:98:15)
    at ClientTCP_transport.connect (C:\Projects\node-opcua\lib\transport\client_tcp_transport.js:134:14)
    at _connect (C:\Projects\node-opcua\lib\client\client_secure_channel_layer.js:665:23)
    at FunctionCall.doCall_ (C:\Projects\node-opcua\node_modules\backoff\lib\function_call.js:156:20)
    at FunctionCall.start (C:\Projects\node-opcua\node_modules\backoff\lib\function_call.js:145:10)
    at _establish_connection (C:\Projects\node-opcua\lib\client\client_secure_channel_layer.js:752:21)
    at ClientSecureChannelLayer.create (C:\Projects\node-opcua\lib\client\client_secure_channel_layer.js:756:5)
    at C:\Projects\node-opcua\lib\client\client_base.js:333:27
    at C:\Projects\node-opcua\node_modules\async\dist\async.js:5285:13
    at replenish (C:\Projects\node-opcua\node_modules\async\dist\async.js:871:21)

I'm not sure my server is OK.　Here is the information of my demo server,

I followed #212 . and applyed the change1.

Then I run the server, There
$ node bin/simple_server
server PID          : 3564

registering server to :opc.tcp://WIN-74C3L8J7J4L:4840/UADiscovery
Ignoring reversed keyword                                               nodeClass
err Cannot find module 'usage'
skipping installation of cpu_usage and memory_usage nodes
  server on port      : 26543
  endpointUrl         : opc.tcp://WIN-74C3L8J7J4L:26543
  serverInfo          :
      applicationUri                  : urn:WIN-74C3L8J7J4L:NodeOPCUA-Server
      productUri                      : NodeOPCUA-Server
      applicationName                 : locale=en text=NodeOPCUA
      $applicationType                : SERVER
      gatewayServerUri                : null
      discoveryProfileUri             : null
      discoveryUrls                   :
      productName                     : NODEOPCUA-SERVER
  buildInfo           :
      productUri                      : NodeOPCUA-Server
      manufacturerName                : Node-OPCUA : MIT Licence ( see http://node-opcua.github.io/)
      productName                     : NODEOPCUA-SERVER
      softwareVersion                 : 0.0.55
      buildNumber                     : 1234
      buildDate                       : Thu Jun 30 2016 10:10:57 GMT+0800 (China Standard Time)

  server now waiting for connections. CTRL+C to stop

lhf552004, 2016-06-30:
I didn't find any solution in the item list.

erossignon, 2016-06-30:
For some reason the server exposes its endpoint url as opc.tcp:\WIN-74C3L8J7J4L:26543
with a backslash instead of a slash : opc.tcp://WIN-74C3L8J7J4L:26543 
parseEndpointURL might not allow this.

closed 2016-06-30 by lhf552004