node-opcua/node-opcua #228: Does the opcua server run as a singleton instance?

michikuehne, 2016-07-06:
Hello,

we discovered an non deterministic behavior in our test suite.
To create an deterministic test scenario we stop and start all related servers for each test.
In one test we simulated a "session limited reached" failure by starting the node-opcua server with maxAllowedSessionNumber = -1. 
It turns out that all following tests fail cause they were also executed with an opcua server with reach session limit even if we create a new server each time.

Therefore I made the assumption that the node-opcua server is running as signleton instance. Is that correct? And if yes, can I change this?

Thanks in advance.

Best,
Michael

erossignon, 2016-07-06:
It should be possible to create multiple servers in one process.
Can you create a gist with a minimalist mocha unit test that hightlight this process so we can investigate it ?

michikuehne, 2016-07-07:
Hi Etienne,

I isolated the described scenario to a single test and could not reproduce the describe behavior:
https://gist.github.com/michikuehne/b44b93f5b057bbb82037357e585202e5

Therefore I guess the error is not related to opcua, as I supposed in the first place.

Best,
Michael

closed 2016-07-07 by michikuehne