node-opcua/node-opcua #229: Support Promisify functions

BLamy, 2016-07-08:
It would be nice to see the api use promises instead of callback functions. 

I haven't exactly narrowed it down 100% but even promisifying functions doesn't seem to work. Likely related to use of deferrer: https://github.com/node-opcua/node-opcua/blob/master/lib/client/client_base.js#L566

With async and await coming to javascript promises are even more advantageous. My only workaround was to create a function which looks like this

```javascript
const asyncApply = (target, fn, ...args) => (
  new Promise((resolve, reject) => {
    fn.call(target, ...args, (e, ...results) => {
      if (e) reject(e);
      else resolve(...results);
    });
  })
);
```

which gives me something a little like this but it feels kind of hacky.
```javascript
async function connect(client, endpoint, user, subscriptionConfig){
      await asyncApply(client, client.connect, endpoint);
      const session = await asyncApply(client, client.createSession, user);
      const subscription = new opcua.ClientSubscription(session, subscriptionConfig);
      subscription
        .on('started', _ => console.log('started'))
        .on('keepalive', _ => console.log('keepalive'))
        .on('terminated', _ => console.log('termianted');
}
```

erossignon, 2016-07-18:
You're right, making the API more Promise friendly would be nice.
However, it seems strange that close session always returns a error, this would have benn spotted by one of the unit test I guess. May be there is a issue elsewhere.

BLamy, 2016-08-13:
I've also begun using RXJS to wrap all monitor calls which seems to be a good fit. If ever considering promises it may also be worth looking at [observables](https://github.com/tc39/proposal-observable). 

A simplistic implementation would look a little something like this
```js
import {
  BehaviorSubject,
  Observable
} from 'rxjs/Rx';

function subscribe(subscription, config) {
  const observable = new BehaviorSubject();
  subscription.monitor(config).on('changed', observable.next.bind(observable));
  return observable;
}

...

const subscription = new opcua.ClientSubscription(session, someConfig);
const node1$ = subscribe(subscription, configForNode1);
const node2$ = subscribe(subscription, configForNode2);
const node1Value$ = node1$.map(node1 => node1.value.value)
const node2Value$ = node2$.map(node2 => node2.value.value)
const total$ = Observable.combineLatest(node1Value$, node2Value$, (node1Value, node2Value) => (
  node1Value + node2Value
);

...

total$.subscribe(
  total => {
    console.log(total);
  }
);
```

erossignon, 2017-01-07:
see @coussej technic to promisify the node-opcua API using bluebird, as described in #291

crussell52, 2017-11-18:
Relatively new to the Node world -- exploring this library in comparison with UA implementations in other languages and the native async/await pattern would greatly simplify implementations and make them much more readable/maintainable.

I have been successful with early attempts at promisifying with `bluebird`, but that has the unfortunate side effect of preventing IDE code completion.

Has there been any progress towards officially supporting promises?

erossignon, 2018-01-29:
node-opcua@0.2.0 onward now  supports promises and async/await pattern 
http://node-opcua.github.io/tutorial/2018/01/22/node-opcua-embraces-async-await.html

closed 2018-01-29 by erossignon