node-opcua/node-opcua #230: Remove Requirish dependency

BLamy, 2016-07-11:
A library like this really seems like it's going to cause more harm than good down the line. It doesn't seem to be very popular, well maintained,l and it looks like it primarily works by injecting nasty side effects into require.

Would you accept a PR if I removed this dependency?

I'm trying to package node-opcua using web pack and I keep getting the following. 

```
ERROR in ./~/node-opcua/index.js
[0] Module not found: Error: Cannot resolve module 'lib/address_space/make_relative_path' in /Users/brett/Projects/calvaryhistorian/node_modules/node-o
pcua
[0]  @ ./~/node-opcua/index.js 87:34-81
[0]
[0] ERROR in ./~/node-opcua/index.js
[0] Module not found: Error: Cannot resolve module 'lib/data_access/EUInformation' in /Users/brett/Projects/calvaryhistorian/node_modules/node-opcua
[0]  @ ./~/node-opcua/index.js 91:48-88 92:48-88
[0]
[0] ERROR in ./~/node-opcua/index.js
[0] Module not found: Error: Cannot resolve module 'lib/client/client_utils' in /Users/brett/Projects/calvaryhistorian/node_modules/node-opcua
[0]  @ ./~/node-opcua/index.js 113:30-64
```

erossignon, 2016-07-18:
The main reason why we use "requirish" is that it simplifies and normalizes the way we use required to include all sources files, regardless of how deep they are in the library file hierarchy .
And so far the side effect you mentioned can be circumvent relatively easily. 
Replacing  "requirish" is possible but not at the expense of making the code more difficult to edit and maintain.
I suggest that we investigate what's wrong with using node-opcua with web-pack, first.
Can you submit a minimalist example (step by step guide) that reproduce the problem you face ?

BLamy, 2016-07-19:
Yes, I'll prepare one

My use case is that I am trying to create an electron application. Therefore, I am directly communicating with node-opcua using react. 

I'm starting with this boilerplate:
https://github.com/chentsulin/electron-react-boilerplate

I'm looking to create elements somewhere along the lines of this:
```javascript
// Element
const CreateReactOpcExampleElement = acync (session, nodeId) =>  {
   const tag =  await asyncCall(session, session.readVariableValue, nodeId);
   return label => (
      <h1>
         <span>{label}</span>
         <span>{tag.value.value}</span>
      </h1>
   );
}

// ...
import opcua from 'node-opcua'; // <--- Will cause error due to webpack.
import { url, opts, user, subscriptionOpts } from './config';
import { nodeId } from './mocks';

try {
   const client = new opcua.OPCUAClient(opts);
   await client.asyncConnect(url);
   const session = await client.asyncCreateSession(user); 
   const OpcElement = CreateReactOpcExampleElement(session, nodeId);
   ReactDOM.render(<OpcElement label="foo" />, document.getElementById('app'));
} catch (e) {
  console.error(e);
  ReactDOM.render(<span>Error: {e}</span>, document.getElementById('app'));
} 

```

## Work around
My only work around is to create a global variable outside of bundle.js
```html
<!DOCTYPE html>
<html>
  <head>
    <meta charset="utf-8">
    <title>Hello Electron React!</title>
    <link href='https://fonts.googleapis.com/css?family=Roboto:400,300,500' rel='stylesheet' type='text/css'>
    <link rel="stylesheet" href="../node_modules/font-awesome/css/font-awesome.min.css" />
    <script>
      // HACK
      // The Node OPC UA library does not play nicely with webpack.
      // Therefore, we expose it as a global from here.
      window.opcua = require('node-opcua');
    </script>
    <script>
      (function() {
        if (!process.env.HOT) {
          const link = document.createElement('link');
          link.rel = 'stylesheet';
          link.href = '../dist/style.css';
          document.write(link.outerHTML);
        }
      }());
    </script>
  </head>
  <body>
    <div id="root"></div>
    <script>
      (function() {
        const script = document.createElement('script');
        script.src = (process.env.HOT) ? 'http://localhost:3000/dist/bundle.js' : '../dist/bundle.js';
        document.write(script.outerHTML);
      }());
    </script>
  </body>
</html>
```
Then I have a class which wraps opcua and imports it as so. 
```javascript
let opcuaImport = window.opcua;
if (!opcuaImport) {
  opcuaImport = require('node-opcua');// eslint-disable-line global-require
}
// ...
const opcua = { ...opcuaImport, ...opcuaExtensions };
export default opcua;
```

erossignon, 2016-07-28:
interesting pointer https://gist.github.com/branneman/8048520

erossignon, 2016-08-07:
@BLamy , let me know if the latest change above fixes the issue.

BLamy, 2016-10-25:
The above fix did take care of my problem. Haven't had any requirish related issues since. 

That said, If you have any plans of migrating to ES6,  [babel-plugin-module-resolver](https://github.com/tleunen/babel-plugin-module-resolver) would be a nice drop-in replacement for requirish. It would patch your paths at build time instead of at runtime.

closed 2016-10-25 by BLamy