node-opcua/node-opcua #231: Server not accepting clients with greater version numbers

cosarara97, 2016-07-12:
When receiving a TCP Hello message, the server does the following check:

```javascript
if (helloMessage.protocolVersion < 0 || helloMessage.protocolVersion > self.protocolVersion) {
            // invalid protocol version requested by client
            self._abortWithError(StatusCodes.BadProtocolVersionUnsupported, "Protocol Version Error" + self.protocolVersion, callback);
}
```
[https://github.com/node-opcua/node-opcua/blob/master/lib/transport/server_tcp_transport.js#L168](url)

The specification, though, says the following:

> The Server shall always accept versions greater than what it supports.

Also, ProtocolVersion should be an UInt32, so the < 0 test seems wrong to me.

erossignon, 2016-07-13:
well spotted. we shall fix this.
Can you tell what "client" did you use to discover the issue ?

cosarara97, 2016-07-13:
I found this when using a client running an older version of node-opcua against a recent node-opcua server (since it used to report itself as using version 1 instead of version 0).

erossignon, 2016-07-14:
This  must be an old version of node-opcua !

closed 2016-07-14 by erossignon