node-opcua/node-opcua #233: ServerStatus is not updating.

awcullen, 2016-07-31:
Our fix for #221 causes a client that is monitoring ServerStatus to not receive published updates.  
The easiest fix is ask the server engine to clone ServerStatus before sending to client.

In lib / server / server_engine.js  line 590: 
'
            bindVariableIfPresent(makeNodeId(VariableIds.Server_ServerStatus), {
                get: function() {
                    var clone= new ServerStatus({
                        startTime: self.serverStatus.startTime,
                        currentTime: new Date(),
                        state: self.serverStatus.state,
                        buildInfo: self.serverStatus.buildInfo,
                        secondsTillShutdown: self.serverStatus.secondsTillShutdown,
                        shutdownReason: self.serverStatus.shutdownReason
                    });
                    return new Variant({
                        dataType: DataType.ExtensionObject,
                        value: clone
                    });
                },
                set: null
            });

'  
The cloned status is properly detected as a datachange by the code in MonitoredItem.

Cheers, Andrew

erossignon, 2016-09-05:
@awcullen , let see how we can make this more efficient and prevent the creation of a new structure each time ```get``` is called. Let's start by writing a unit test to highlight the wrong behavior.

What's about this scenario ?

**Given** a client , connected to our server
**When** we create a subscription with a monitoredItem on the ServerStatus 
**Then** we should received a Notification  changed on a regular basis 

The notification shall be raised because serverStatus.currentDate is supposed to be frequently updated.

awcullen, 2016-09-05:
I don't know much about testing. 
```
    it("a subscription should report monitored item notification for ServerStatus", function (done) {

        var subscription = new Subscription({
            publishingInterval: 1000,
            maxKeepAliveCount: 20,
            publishEngine: fake_publish_engine
        });

        subscription.on("monitoredItem", function (monitoredItem) {
            monitoredItem.samplingFunc = install_spying_samplingFunc();
        });

        var monitoredItemCreateRequest = new MonitoredItemCreateRequest({
            itemToMonitor: {nodeId: 'i=2256' },
            monitoringMode: subscription_service.MonitoringMode.Reporting,
            requestedParameters: {
                queueSize: 1,
                samplingInterval: 1000
            }
        });

        var monitoredItemCreateResult = subscription.createMonitoredItem(addressSpace, TimestampsToReturn.Both, monitoredItemCreateRequest);
        monitoredItemCreateResult.statusCode.should.eql(StatusCodes.Good);

        var monitoredItem = subscription.getMonitoredItem(monitoredItemCreateResult.monitoredItemId);

        // data collection is done asynchronously => let give some time for this to happen
        this.clock.tick(5);

        // at first, _collectNotificationData  should has 1 notification with current dataItem value
        var notifications = subscription._collectNotificationData();
        should(notifications.length).equal(1);

        // now simulate some time passes
        this.clock.tick(5000);

        // then, _collectNotificationData  should collect at least 5 values
        notifications = subscription._collectNotificationData();
        notifications.length.should.eql(5);

        subscription.on("terminated", function () {
            done();
        });
        subscription.terminate();

    });
```

erossignon, 2016-09-05:
Thank you Andrew. The test you propose uses low level code. We could also create a end_to_end test that directly use the public API. Let me propose this to you.

closed 2016-09-15 by awcullen