node-opcua/node-opcua #234: Does the client support connecting through a proxy?

vrecer, 2016-08-01:
Hello,
I just wanted to ask if the latest version of node-opcua supports client connections through a proxy. 
And if it does I would like to know how this can be set up in the code, because I didn't find any references in the client_base.js. 
If it's still missing can you let me know if you're planning to add it to the client?

erossignon, 2016-12-06:
No we don't. I am not sure the OPCUA spec is providing anything related to using proxy with OPCUA.
Can you describe the network architecture you want to deal with ?

vrecer, 2016-12-06:
Hey, thanks for an answer. I have a situation where I would have to connect to a remote network. The possibilities would be setting up a vpn or to connect through a proxy, and it would be great to also connect through a proxy not only with a vpn.

erossignon, 2016-12-06:
The vpn is probably the best option. OPCUA doesn't like "man in the middle"<- proxy and a secure connection would fail.

vrecer, 2016-12-06:
Ok thanks for the information, I'll stick to vpn then.

closed 2017-01-07 by erossignon