node-opcua/node-opcua #235: 0.0.56 postinstall cert generation error on osx

reijosirila, 2016-08-02:
------------------------------------------------------------------- 

      set OPENSSL_CONF=/Users/reijo/dev/app/api_test/node_modules/node-opcua/node_modules/node-opcua-pki/lib/pki/empty_config.cnf
              OPENSSL_CONF /Users/reijo/dev/app/api_test/node_modules/node-opcua/node_modules/node-opcua-pki/lib/pki/empty_config.cnf
              RANDFILE     /Users/reijo/dev/app/api_test/certificates/PKI/own/private/random.rnd
              CMD         openssl  ca  -config conf/caconfig.cnf  -startdate 20160801151012Z -enddate 20170801151012Z -batch -out "/Users/reijo/dev/app/api_test/certificates/client_cert_1024.pem" -in "/Users/reijo/dev/app/api_test/certificates/client_cert_1024.pem.csr"
              CWD          /Users/reijo/dev/app/api_test/certificates/CA
    stderr Using configuration from conf/caconfig.cnf
    stderr start date is invalid, it should be YYMMDDHHMMSSZ
    stderr 73805:error:0E06D06C:configuration file routines:NCONF_get_string:no value:/BuildRoot/Library/Caches/com.apple.xbs/Sources/OpenSSL098/OpenSSL098-59.40.2/src/crypto/conf/conf_lib.c:329:group=CA_default name=email_in_dn
########################################### OPENSSL ERROR  ###############################
Command failed: /bin/sh -c "openssl" ca  -config conf/caconfig.cnf  -startdate 20160801151012Z -enddate 20170801151012Z -batch -out "/Users/reijo/dev/app/api_test/certificates/client_cert_1024.pem" -in "/Users/reijo/dev/app/api_test/certificates/client_cert_1024.pem.csr"
Using configuration from conf/caconfig.cnf
start date is invalid, it should be YYMMDDHHMMSSZ
73805:error:0E06D06C:configuration file routines:NCONF_get_string:no value:/BuildRoot/Library/Caches/com.apple.xbs/Sources/OpenSSL098/OpenSSL098-59.40.2/src/crypto/conf/conf_lib.c:329:group=CA_default name=email_in_dn

########################################### OPENSSL ERROR  ###############################
ERROR  Command failed: /bin/sh -c "openssl" ca  -config conf/caconfig.cnf  -startdate 20160801151012Z -enddate 20170801151012Z -batch -out "/Users/reijo/dev/app/api_test/certificates/client_cert_1024.pem" -in "/Users/reijo/dev/app/api_test/certificates/client_cert_1024.pem.csr"
Using configuration from conf/caconfig.cnf
start date is invalid, it should be YYMMDDHHMMSSZ
73805:error:0E06D06C:configuration file routines:NCONF_get_string:no value:/BuildRoot/Library/Caches/com.apple.xbs/Sources/OpenSSL098/OpenSSL098-59.40.2/src/crypto/conf/conf_lib.c:329:group=CA_default name=email_in_dn

Command failed: /bin/sh -c "openssl" ca  -config conf/caconfig.cnf  -startdate 20160801151012Z -enddate 20170801151012Z -batch -out "/Users/reijo/dev/app/api_test/certificates/client_cert_1024.pem" -in "/Users/reijo/dev/app/api_test/certificates/client_cert_1024.pem.csr"
Using configuration from conf/caconfig.cnf
start date is invalid, it should be YYMMDDHHMMSSZ
73805:error:0E06D06C:configuration file routines:NCONF_get_string:no value:/BuildRoot/Library/Caches/com.apple.xbs/Sources/OpenSSL098/OpenSSL098-59.40.2/src/crypto/conf/conf_lib.c:329:group=CA_default name=email_in_dn

erossignon, 2016-08-03:
Can you update openssl to the latest version (> 1.0 )and let me know if this fixes the issue.
I would rather add a prerequisite to use a recent version of openssl than forcing  certifictate year on 2 digits only.
http://stackoverflow.com/questions/15185661/update-openssl-on-os-x-with-homebrew
http://apple.stackexchange.com/questions/126830/how-to-upgrade-openssl-in-os-x

reijosirila, 2016-08-03:
Yes, it works with newer openssl.

makhtardiouf, 2016-08-18:
Hi,

I bumped on the cerficate error, and running "**npm test**"  generated the missing certificates.