node-opcua/node-opcua #236: Returning to the console prompt after shutdown

c106676, 2016-08-03:
How is the recommended way to shutdown the server and return to the console prompt?
In the `sample_server.js` I tried adding 
```
console.log("Shutting down in 10 seconds.");
server.shutdown(10000, function(err){
	if(err){
		console.log("Error while shutting down the server: '", err, "'");
	}else{
		console.log("Successfully shut down the server.")
	}
});
```
after the `server.initilialize()`. The server shuts down, but the program is not ending. It has to be ended by Ctrl+C explicitly.

erossignon, 2016-08-03:
This is the expected behavior.  
  - click CTRL+C and wait for the server to properly shutdown itself.
  - alternatively you can send a kill -USR signal to the nodejs process running your server.

A server is supposed to work until we stop it. OPCUA doesn't specify a way to stop a server from a opcua client side.
What would be your requirement ? Please, expand your though.

erossignon, 2016-08-03:
I misread your request . In your case I would try:
```
    server.shutdown(1000, function () {
        process.exit(-1);  //  add this line to stop the process
    });
```

c106676, 2016-08-03:
`process.exit(-1)` returns to the command prompt indeed and is sufficient for my current case. But one should really be certain not to have any pending tasks:
```
setTimeout(function(){
    console.log("Will I get printed?");
}, 1001);
server.shutdown(1000, function(){
    process.exit(-1);
});
```

closed 2017-04-19 by c106676