node-opcua/node-opcua #237: reconnecton problems

reijosirila, 2016-08-03:
I did some reconnection tests if using monitored items. 
In my case reconnection fails: 
1) If I remote default route and put it back. Console output: 
```
ClientTCP_transport Socket Error read ECONNRESET
Debug: started reconnect
Connect Tore ::: Test PLC@opc.tcp://192.168.114.93:4840
Connect Tore ::: Test PLC@opc.tcp://192.168.114.93:4840
Connect Tore ::: Test PLC@opc.tcp://192.168.114.93:4840
OPC UA connection_reestablished
Debug: Reconnected
err =  { [Error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2016-08-03T14:06:23.354Z
   requestHandle               /* IntegerId                        */: 2               0x2
   serviceResult               /* StatusCode                       */: BadSessionIdInvalid (0x80250000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};]
  response: 
   ServiceFault {
     responseHeader: 
      ResponseHeader {
        timeStamp: Wed Aug 03 2016 17:06:23 GMT+0300 (EEST),
        requestHandle: 2,
        serviceResult: [Object],
        serviceDiagnostics: [Object],
        stringTable: [Object],
        additionalHeader: null } } }
warning : disconnection : closing pending sessions
OPC UA Closed
```

2) If I disable / enable network interface, then connection timeout occurs and reconnection does not start at all. Console output
```
Timeout .... waiting for response for  { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=12238
 timeStamp                     /* UtcTime                          */: 2016-08-03T14:04:54.627Z
 requestHandle                 /* IntegerId                        */: 29               0x1d
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: 
 timeoutHint                   /* UInt32                           */: 50000               0xc350
 additionalHeader              /* ExtensionObject                  */: null
};
Debug: OPC UA timed_out_request
 Timeout .... waiting for response for  { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=12238
 timeStamp                     /* UtcTime                          */: 2016-08-03T14:04:59.756Z
 requestHandle                 /* IntegerId                        */: 30               0x1e
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: 
 timeoutHint                   /* UInt32                           */: 50000               0xc350
 additionalHeader              /* ExtensionObject                  */: null
};
Debug: OPC UA timed_out_request
Connect Tore ::: Test PLC@opc.tcp://192.168.114.93:4840
 Timeout .... waiting for response for  { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=12238
 timeStamp                     /* UtcTime                          */: 2016-08-03T14:05:04.896Z
 requestHandle                 /* IntegerId                        */: 31               0x1f
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: 
 timeoutHint                   /* UInt32                           */: 50000               0xc350
 additionalHeader              /* ExtensionObject                  */: null
};
Debug: OPC UA timed_out_request
 Timeout .... waiting for response for  { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=12238
 timeStamp                     /* UtcTime                          */: 2016-08-03T14:05:10.154Z
 requestHandle                 /* IntegerId                        */: 32               0x20
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: 
 timeoutHint                   /* UInt32                           */: 50000               0xc350
 additionalHeader              /* ExtensionObject                  */: null
};
/..../
```

Using:
```
var OPCSubscriptionParams = {
	requestedPublishingInterval: 100,//,
	requestedLifetimeCount: 1000, //1000,
	requestedMaxKeepAliveCount: 12,//12,
	maxNotificationsPerPublish: 10, //10,
	publishingEnabled: true,
	priority: 10
};
```

erossignon, 2016-09-06:
This needs to be investigated.

erossignon, 2016-10-28:
Can you provide more information about the server and the client in your case.
What is running under opc.tcp://192.168.114.93:4840 ?

mepdr, 2016-12-19:
Sometimes I got the error below using kepware server
```
 Timeout .... waiting for response for  PublishRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=1028567098
 timeStamp                     /* UtcTime                          */: 2016-12-19T10:00:01.948Z
 requestHandle                 /* IntegerId                        */: 124               0x7c
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */:
 timeoutHint                   /* UInt32                           */: 100000               0x186a0
 additionalHeader              /* ExtensionObject                  */: null
};
```

CrNix, 2017-03-27:
Hi, is OPC UA client supposed to support automatic reconnection with monitored items?

I have a subscription with some monitored items. And when I disconnect and connect again (by removing the ethernet cable from router and plugging in back in) the client reconnects OK and my additional OPC UA calls to read nodes succeed.

But all monitored items are terminated and values are not published anymore.

Is there a way to preserve publishing or should I reinitialize the subscription somehow?

Thank you

erossignon, 2017-03-27:
Yes, this is probably because your subscription has timed out. You need to extend the subscription livetime count to a very high value. Please remember that while the client and server are disconnected, the data will accumulate on the server side in this case to be delivered when the connection is reestablished.
Please refer to the unit tests to see how this is implemented and supported.

CrNix, 2017-03-27:
@erossignon That one works. Thank you very much. Also the number 0 works. It works like infinite?

dushyantbangal, 2017-03-31:
@erossignon 
Hi, I'm getting following log after I keep the monitors running for some time.
Is it the same issue?

```
Timeout .... waiting for response for  ReadRequest { /*RequestHeader*/
authenticationToken           /* NodeId                           */: ns=0;i=22
timeStamp                     /* UtcTime                          */: 2017-03-28T17:09:37.934Z
requestHandle                 /* IntegerId                        */: 315               0x13b
returnDiagnostics             /* UInt32                           */: 0               0x0
auditEntryId                  /* UAString                         */:
timeoutHint                   /* UInt32                           */: 0               0x0
additionalHeader              /* ExtensionObject                  */: null
};
warning :  Transaction has timed out
Timeout .... waiting for response for  ReadRequest { /*RequestHeader*/
authenticationToken           /* NodeId                           */: ns=0;i=22
timeStamp                     /* UtcTime                          */: 2017-03-28T17:09:51.268Z
requestHandle                 /* IntegerId                        */: 316               0x13c
returnDiagnostics             /* UInt32                           */: 0               0x0
auditEntryId                  /* UAString                         */:
timeoutHint                   /* UInt32                           */: 0               0x0
additionalHeader              /* ExtensionObject                  */: null
};
```

reijosirila, 2018-04-16:
for me 0.3.0 fixes issue

erossignon, 2018-04-16:
Great.

closed 2018-04-16 by erossignon