node-opcua/node-opcua #240: Facing Problems while Connecting to Siemens S7-1200

sagardubal, 2016-08-12:
I am having problems try to connect to our OPC Server (S7-1200 Siemens PLC), We are able to connect to demo server perfectly. Can you help, regarding the connection process for the Siemens PLC.

Lobo-T, 2016-08-12:
The S7-1200 does not have an OPC server in itself.  You will need a computer running the Simatic NET OPC server.

sagardubal, 2016-08-12:
We are able to communicate the 1200 PLC on Markitron Open Automation Software, so i believe we should be able to communicate on the Node.JS, We have gone through all your code and tutorial but somehow we feel we are going wrong in some place in connecting the PLC. 

Steps we have done, 
Tried connecting to the Demo Client, Worked Successfully. 
Tried Connecting to the PLC on its IP with port 4845,4860,4861 
But we were not able to connect. 

Kindly Suggest what can we do.

Other Hardware Available : Siemens S7 ET200S

Lobo-T, 2016-08-14:
Maybe I'm not understanding exactly what you are trying to do.

Are you trying to use Node OPC UA to communicate directly with a S7-1200 PLC?
As I said, this is not possible, the S7 does not contain an OPC server.

You mention that you have communicated with the S7 with "Markitron Open Automation Software", I can not find a product with this exact name.  But Matrikon sells an OPC server that can communicate with the S7 over the S7 communications protocol.   And you can then communicate with the Matrikon server over OPC.  I have never used the Matrikon servers so I'm not sure if they suppert OPC UA or only DCOM.

hsch0, 2016-08-19:
Thats interesting. I was also looking for a solution to communicate with a Siemens PLC and thought that OPC is already integrated in the PLC. Does other PLC manufactures have an OPC server builtin?

Lobo-T, 2016-08-21:
The only PLC manufacturer that I know of that has the option of running an OPC server on the PLC is Beckhoff.  But their PLCs are more like industrial computers running Windows CE.

hsch0, 2016-08-23:
and there is currently no open source OPC server with S7 communication included? By using the Libnodave library it would be actually possible I guess...

Lobo-T, 2016-08-23:
For someone using Siemens equipment the price of the Simatic OPC server is in most cases probably negligible.  But I can see that someone would want to avoid using a Windows PC.

I have found a Node.js S7 protocol library, but I haven't tested it: https://github.com/plcpeople/nodeS7
That combined with Node OPC-UA should be a solution.

sagardubal, 2016-08-29:
Will try the  S7 protocol library and get back to you. Meanwhile, how can we directly communicate with PLC on OPC-UA (Node.js) without use of any Windows PC. We were communicating the Simens PLC  from https://www.openautomationsoftware.com/downloads/open-automation-software/

mram888, 2016-09-08:
I have tried the S7 protocol library @Lobo-T said and I am having some problems with my S7-1200 writting and reading.
[image]

plcpeople, 2016-09-09:
I am the author of that package, if you post some more details on what is going on I will look into it.  Just from the screenshot above it looks like you may be trying to read the same area of memory both as an integer and a floating point number.  What does the structure of DB1 really look like?  I don't have an S7-1200 but I have heard when reading/writing with the S7 protocol with that PLC you can get errors if the data type you are using doesn't match the structure of the data block.