node-opcua/node-opcua #241: 0.0.57 - Error Certificate file must exist - install was OK

biancode, 2016-08-14:
AssertionError: fs.existsSync(this.certificateFile), "Certificate file must exist :" + this.certificateFile

[image]

biancode, 2016-08-14:
Executing  node bin/generate_opcua_classes.js --clear 
done ... (0)
Executing  node bin/crypto_create_CA.js demo
done ... (0)

erossignon, 2016-08-14:
Are you on Mac ? may be https://github.com/node-opcua/node-opcua/issues/235 could help

biancode, 2016-08-14:
OK, that was also the openssl problem and brew force link doesn't work

steps done on Mac OS X:
 
- clone from git://git.openssl.org/openssl.git
- cd into path of that git clone of openssl
- ./Configure darwin64-x86_64-cc shared enable-ec_nistp_64_gcc_128 no-ssl2 no-ssl3 no-comp -openssldir=/usr/local/ssl/macos-x86_64
- make depend
- sudo make install
- uninstall node-opcua package
- install node-opcua package

works ...

see https://wiki.openssl.org/index.php/Compilation_and_Installation

biancode, 2016-08-14:
[image]

closed 2016-08-21 by biancode