node-opcua/node-opcua #243: How to read my reference node's variable under source node

lhf552004, 2016-08-24:
Hi , I have create a reference instance of 'HasChild' reference type at a source node. But How can I access the reference node's variable under the source node?
The structure is below:
|--Element
       |--A-1006-MXZ01 
                |--StCode
        
|--Unit
       |--A-1006
               |--Elements
                        |--A-1006-MXZ01

Here is my unit 'A-1006',it is source node. I have add a reference node 'A-1006-MXZ01'  which is referenced to the node 'A-1006-MXZ01' in the 'Element' folder.
The question is when I get my unit 'A-1006', how can I read the value of 'StCode' of element 'A-1006-MXZ01'?

erossignon, 2016-08-24:
Could you provide a gist with a minimalist workable sample code ?

lhf552004, 2016-08-25:
Sorry, It's my fault. I have created a element node,its nodeId is "ns=1;s=Element.BeltMonitor.=A-0003-BZA20", Meanwhile, I have created a unit node , its nodeId is "ns=1;s=Unit.LBEA.=A-0003": 
```
var elementNode = addressSpace.addFolder(parentNode, {
                                nodeId: elementNodeId,
                                browseName: elementName
                            });
var unitNode = addressSpace.addFolder(parentNode, {
                                nodeId: unitNodeId,
                                browseName: unitName
                            });
var elements = addressSpace.addFolder(unitNode , {
                                nodeId: unitNodeId + '.Elements',
                                browseName: 'Elements'
                            });
var reference ={
                         referenceType: 'HasChild',
                         isForward : true,
                         nodeId :elementNodeId
                          };
 elements .addReference(reference);
```
Now I want  to read or write elementNode when I get the unitNode Id. Actually, I find the nodeId "ns=1;s=Unit.LBEA.Elements.=A-0003.=A-003-BZA20" dosen't exist.
But in the address space, it is in the 'Elements' folder. 
[image]
When I clicked the element, Its Identifier is  "PLC1.Element.BeltMonitor.=A-0003-BZA20"
[image]

lhf552004, 2016-08-26:
Sorry, maybe my words are not clear to understand, I mean, I want to read reference node by unit node.

erossignon, 2018-10-28:
closing due to lack of activity since 2 years

closed 2018-10-28 by erossignon