node-opcua/node-opcua #244: Path character limit exceeded

AlexPeitz, 2016-08-26:
Hello,

I just tried to install the node-opcua package. I am currently developing using Visual Studio 2015 and the NodeJs tools for VS. But every time I try to use the package in my project, I get a build error saying some paths exceed the 260 character limits. I tried to create a new project in a very short path already ("C:\s\") but that did not completely solve the issue. Is there any way to solve this? 

Regards,
Alex

erossignon, 2016-08-30:
Which version of nodejs/npm  are you using ? 
try the most recent one, they manage to flatten the node_modules folder .

AlexPeitz, 2016-08-30:
I'm using node v0.12 right now, and cannot upgrade this due to compatibility reasons with other software. I'll upgrade my npm though. For now I solved it by installing the flatten-packages module and installing the newest windows update which removes the 360 character path limit. Thank you for your reply!

closed 2016-08-30 by erossignon