node-opcua/node-opcua #245: Problem creating EventNotifier monitoring

vrecer, 2016-08-29:
I'm trying to create an EventNotifier subscription for monitoring events of the sample server from the OPC Foundation.
But I'm having problems when using the example that I found in the client 

This subscription is supposed to monitor the simulation node which I know generates some events on starting, stopping etc:
```
      var filter =  new opcua.subscription_service.EventFilter({
     selectClauses: [
              { browsePath: [ {name: 'ActiveState'  }, {name: 'id'}  ]},
              { browsePath: [ {name: 'ConditionName'}                ]}
     ],
     whereClause: []
   });

       var monitoredItem  = UAservers[0].the_subscription.monitor(
      // itemToMonitor:
      {
        nodeId: opcua.resolveNodeId("ns=5;i=31"),
        attributeId: opcua.AttributeIds.EventNotifier,
        indexRange: null,
        dataEncoding: { namespaceIndex: 0, name: null }
      },
      // requestedParameters:
      {
         clientHandle: 12,
         samplingInterval: 3000,
 
         filter: filter,
 
         queueSize: 1,
         discardOldest: true
      },
      opcua.read_service.TimestampsToReturn.Neither
    );

       monitoredItem.on("changed",function(dataValue){
          logger.log(logger.INFO,monitoredItem.itemToMonitor.nodeId + ": " + dataValue);
       });
```

But the problem is that the node does not get monitored, I also see that the item count in the subscription on the server does not increase, like it does if I try monitoring a value.

While monitoring values works without problems with the following code:
  ```
 var test = UAservers[0].the_subscription.monitor(
      // itemToMonitor:
      {
        nodeId: "ns=5;i=11",
        attributeId: opcua.AttributeIds.Value,
        indexRange: null,
        dataEncoding: { namespaceIndex: 0, name: null }
      },
      // requestedParameters:
      {
         clientHandle: 13,
         samplingInterval: 3000,
         filter: null,
         queueSize: 1,
         discardOldest: true
      },
      opcua.read_service.TimestampsToReturn.Neither
    );
	test.on("changed",function(dataValue){
          logger.log(logger.INFO,monitoredItem.itemToMonitor.nodeId + ": " + dataValue.value.value);
       });
	
```

The example I found in client_subsription.js was this:

```
 * Monitoring an Event
 * -------------------
 *
 *  If the monitor attributeId is EventNotifier then the filter must be specified
 *
 * @example:
 *
 *  var filter =  new subscription_service.EventFilter({
 *    selectClauses: [
 *             { browsePath: [ {name: 'ActiveState'  }, {name: 'id'}  ]},
 *             { browsePath: [ {name: 'ConditionName'}                ]}
 *    ],
 *    whereClause: []
 *  });
 *
 *  clientSubscription.monitor(
 *     // itemToMonitor:
 *     {
 *       nodeId: "ns=0;i=2258",
 *       attributeId: AttributeIds.EventNotifier,
 *       indexRange: null,
 *       dataEncoding: { namespaceIndex: 0, name: null }
 *     },
 *     // requestedParameters:
 *     {
 *        clientHandle: 13,
 *        samplingInterval: 3000,
 *
 *        filter: filter,
 *
 *        queueSize: 1,
 *        discardOldest: true
 *     },
 *     TimestampToReturn.Neither
 *   );

```

Can you help by pointing me in the right direction of what I'm doing wrong  or if this is actually not working when trying to subscribe to events?

Thanks!

erossignon, 2016-09-05:
do you mean the alarm & condition demo ?"A&C server" (opc.tcp://localhost:62544/Quickstarts/AlarmConditionServer).
Which node are you trying to monitor for Event ?
[image]

vrecer, 2016-09-06:
I've found the node-opcua sample client and saw how the filter should be created, I guess I was doing something wrong but when I changed to one similar that I found in sample client it now works with the following code:

```
        var fields = [
            "EventId",
            "EventType",
            "SourceNode",
            "SourceName",
            "Time",
            "ReceiveTime",
            "Message",
            "Severity"
        ];

        var eventFilter = opcua.constructEventFilter(fields);

        
        var event_monitoringItem = self.the_subscription.monitor(
        {
            nodeId: "ns=4;i=1287",
            attributeId: opcua.AttributeIds.EventNotifier
        },
        {
            queueSize: 100000,
            filter: eventFilter,
            discardOldest: true
        }
```

I was trying to monitor node "Simulation" on OPC Foundation Sample Server, because that is the only one where I was able to find events on that sample server. 

Thanks for help!

Br,
Vrecer

closed 2016-09-06 by erossignon