node-opcua/node-opcua #248: Dependency on collections modul breaks down Array's find method

mariocytr, 2016-09-02:
Working with the node-opcua I encountered a problem with the Array.find method. It is broken. The collections modul which node-opcua depends on makes a polyfill with the wrong implemented find. It gives back an index instead of an array element.

erossignon, 2016-09-02:
Strange ! Can you investigate and find the offending code ?

erossignon, 2016-09-02:
A grep in the code shows  that the "collection" npm module is reimplementing the Array#find method
```
node_modules/collections/sorted-array.js:SortedArray.prototype.find = function (value, equals, index) {
node_modules/collections/sorted-array.js:SortedArray.prototype.findLast = function (value, equals, index) {
```

BLamy, 2016-10-25:
I can confirm this issue. 

[image]
[image]
[image]
[image]
[image]

erossignon, 2016-10-28:
thank you @BLamy ,
This seems to be a known issue of the ```collections``` package as stated in "http://www.collectionsjs.com/method/find-value" 

> In version 1, this method is called find, which conflicts with the definition of find provided by ECMAScript 6. In version 2, this method is called findValue to eliminate the conflict.

Let's remove the dependency to ```collections``` all together.

BLamy, 2016-11-10:
Any chance you can push a new minor release. This bug keeps coming back to burn me over and over.

erossignon, 2016-11-11:
Now on v0.0.59

closed 2016-11-11 by erossignon