node-opcua/node-opcua #249: Create a new variable in Kepserver using node-opcua

mram888, 2016-09-06:
I have a Siemens 1200 PLC. Using node-opcua client and Kepserver I am able to read the variables and to change the values. Now I would like to create a new variable in the PLC from node-opcua in KepServer.
[image]

I've tried to use node-opcua server because in the examples I've seen how to create variables, but I get an error because I am trying to connect to the same port that KepServer does.

```
var server = new opcua.OPCUAServer({
    port: 49320, // the port of the listening socket of the server
    resourcePath: "", // this path will be added to the endpoint resource name
     buildInfo : {
        productName: "MySampleServer1",
        buildNumber: "7658",
        buildDate: new Date(2014,5,2)
    }
});
```

[image]

How could I deal to create a new variable? and to create a group tag from node-opcua in Kepserver?
Could I instantiate my server in any way and be able to use the opcua server functions?

antmag, 2016-09-15:
Hey erossignon,

Unfortunately I cant fix your issue but I would like to read and write the values of the OPCua server. How you did that?

Thanks in advance,

Antoine

mininggold, 2016-11-01:
You could use Kepware KEPServerEX as a transparent tunnel using:
[https://www.kepware.com/en-us/products/kepserverex/drivers/opc-ua-client/](url)
See the following document:
[https://www.kepware.com/en-us/support/resource-library/connectivity-guides/kepserverex-opc-tunnel/](url)
on page 4 under section 5.1.3 point 18 and the Tip section mentions: 
"Without importing tags, users can address OPC items in the target OPC DA server across the KEPServerEX OPC tunnel by using a particular tag address syntax from the client application connected to the tunnel client"
example:
MyOPCUAchannel1.myOPCUAdevice1.ns=2;s=myOPCDAchannel2.myOPCDAdevice2.tag

"This is called a dynamic tag. Dynamic tags are a benefit when working with an OPC tunnel because the tag is only created in one place"

erossignon, 2018-10-28:
closing due to lack of activity since 2 years

closed 2018-10-28 by erossignon