node-opcua/node-opcua #250: Kepware KEPServerEX OPC UA server blocks connections

mininggold, 2016-09-12:
After a successful connection, a new session and a read(), then close session and close connection, then next sequence of connect, session then read(), it will fail, and it seems to be intermittent. There is one second between reads. Leaving the connection or session open with a read() of multiple nodes causes KepServerEX to lock up to where the KepServerEX exe process has to be killed, for new connections to be accepted. 

Kepware technical support has been contacted and they are not sure how to resolve this issue. 

Each read is a different tag or node and there are almost 2000 tags that are being polled from a web browser. The app uses Node JS express HTTP server with socket.io websockets. OPC UA Subscription seems complex since not all 2000 nodes are added at the same time, but rather on demand, as the browser requests them. Can subscription be changed (add remove nodes) on the fly without closing the session and reconnecting?

There don't seem to be examples available for this scenario with multiple dynamic tags; dynamic meaning adding or removing them from being read on the fly. Most examples are a singe read of a single tag.

This is an awesome library. Thank you for the support.

erossignon, 2018-01-31:
>> Can a subscription be changed (nodes added or deleted) on the fly, and would this be a better solution rather than polling a single node or tag?
yes, full subscription support is now in node-opcua, you can add and delete monitored.

closed 2018-01-31 by erossignon

BartWitdouckVDW, 2019-08-22:
Hello @ghost

I'm having similar problems with KEPServerEX V5, did you find a solution for this problem?
Thanks