node-opcua/node-opcua #252: Read and write values on OPCua server

antmag, 2016-09-15:
Hey everybody,

I created a dynamic dashboard with npm in order to display a lot of values. I created an OPCUA server on my raspberry and sensors are plug in the raspberry. I can read sensors values with codesys or UAexpert when I am connect to the server. Now, I would like to use node-opcua module to read OPCUA server values.
I have got differents files: app.js index.html and script.js
I would like to access to OPCUA values in the file script.js but when it is not possible to do a opcua=require("node-opcua") in the browser file...
Do you have an issue to fix the problem and to read and write on the opcua server?

mininggold, 2016-09-17:
Hey, this is a fantastic project. 
Node-opcua is not really compatible with the browser as far as I know. Check out this project for ideas: 
[https://github.com/node-opcua/node-opcua-htmlpanel]
To read values from the browser you need socket.io as a web socket, which then means you need a web server.  I would really like know if you find a better way, just post your solution here.

antmag, 2016-09-27:
Hey,

Indeed, I used socket.io to read the values on the server. It is very easy to do that and the [https://github.com/node-opcua/node-opcua-htmlpanel] project is perfect to learn how you can do that.
Now, I am searching and finiding a lot of things to change the variables values directly on the server with a checkbox on my dashboard for ecample.

If anybody have an idea...

erossignon, 2018-10-28:
closing due to lack of activity since 2 years

closed 2018-10-28 by erossignon