node-opcua/node-opcua #253: Implementing RESTful node-opcua API

AlexPeitz, 2016-09-27:
Hello Etienne,

I am using your code to implement a RESTful API on nodejs/express basis. Basically, what I want to do is connect to a server, get all available Methods from that Server and return these to the client. The client is then able to call these methods. Therefore, I want to give the client the data about the method (id, browseName etc) but also additional data like a list of InputArguments and their datatypes for example. I add these arguments manually to my objects, and then try to return an array of these "custom" method objects to the client using express. The object then gets stringified, but there my custom data is lost. 

Basically, here is my object before sending: 

```
{
$nodeClass: Object,
Args: Array,
browseName: Object,
displayName: Object,
isForward: True,
nodeId: Object,
parentNodeId: Object,
referenceTypeId: Object,
typeDefinition: Object
}
```

Nowm parentNodeId is the ID from the parent object and Args is an Array of the InputArguments, where each Array element is composed of the Argument's name and DataType, obtained when using the session.read() function. 

Now, when I want to send this over to the client, this object gets stringified and it looks like this:

```
{
"referenceTypeId": "ns=0;i=47"
"isForward": true
"nodeId": "ns=10;i=14"
"browseName": {
"namespaceIndex": 10
"name": "OpenCloseDoor"
}-
"displayName": {
"text": "OpenCloseDoor"
}-
"nodeClass": "Method"
"typeDefinition": "ns=10;i=420"
}
```

so I lose my custom data, the IDs get formatted to these strings etc. But I don't want that, how can I prevent it? 

Regards,
Alex

sanderd17, 2017-03-07:
Do you use JSON.stringify to stringify the data?

I would like to see your code though, I'm just starting with OPC-UA on Node, and get a bit lost on how to browse through nodes (how to create a valid path etc).

AlexPeitz, 2017-03-16:
I am using the express Framework to send back the http answers. Somehow though, this issue got fixed and now I get nice objects, for example like this: 

```javascript
{
    methods: [
    {
        browseName: "OpenCloseDoor",
        id: "ns=10;i=14",
        nodeClass: "Method",
        parent: "ns=10;i=1",
        inputArguments: [
        {
            name: "OpenCloseDoor",
            dataType: "Boolean",
            valueRank: -1,
            arrayDimensions: [],
            description: {}
        }],
        outputArguments: [],
    }],
}
```

Okay so my code is packed into different REST calls. I'll post them all here so you can see a bit how I did it. First, here are the variables I use
```javascript
var express = require('express');
var opcua = require('node-opcua');
var async = require('async');
var bodyParser = require('body-parser');

//Initialize express
var app = express();
var expressPort = 3000;

//Initialize OPC Variables
var client = new opcua.OPCUAClient;
var server = new opcua.OPCUAServer;

var AttributeIds = opcua.AttributeIds;
var DataTypes = opcua.DataType;
var StatusCodes = opcua.StatusCodes;
var the_session;
var allMethods, allNodes, allObjects, allVariables;

app.use(express.static(__dirname));
app.use(bodyParser.json());       // to support JSON-encoded bodies
app.use(bodyParser.urlencoded({
    // to support URL-encoded bodies
    extended: true
}));
```

Now the connect function, it is a POST request with in the body the endpointURL of the server: 

```javascript
app.post('/Connect', function (req, res) {
    endpointUrl = req.body.endpointURL;
    async.series([
        // step 1 : disconnect if already a connection
        function (callback) {
            client.disconnect(function (err) {
                callback(err);
            });
        },
        //step 2: Connect to the server
        function (callback) {
            client = new opcua.OPCUAClient;
            client.keepSessionAlive = true;
            client.ApplicationName = "NodeOPCUAClient";
            client.clientName = "NodeJSClient";
[... 340 lines cut ...]
        for (var j = 0; j < arguments[i].length; j++) {
            arg.push(new opcua.Variant({ dataType: opcua.DataType[arguments[i][j].dataType], value: arguments[i][j].value }));
            //valueArray.push(new opcua.Variant({ value: inputArgs[i][j].value, dataType: 11 }));
        }
        //arg.push(new opcua.Variant({dataType: 11, value: valueArray, dimension: 9}));
        methodsToCall.push({ objectId: parentIds[i], methodId: methodIds[i], inputArguments: arg });
    }

    the_session.call(methodsToCall, function (err, response) {
        console.log(err);
        res.send(response);
    })
});
```

After all those GET and POST requests, you have to initialize the express server:

```javascript
//Initialize the server to listen on a custom Port
app.listen(expressPort, function () {
    console.log('server listening on port 3000');
});
```

I hope this will help you a bit!

closed 2019-08-03 by stale[bot]