node-opcua/node-opcua #254: Error: cannot locate certificate file node_modules\node-opcua\certificates\client_selfsigned_cert_102 4.pem

luizgppires, 2016-10-05:
Install was ok. 
node-opcua version: 0.0.57

What I get:

`node_modules\node-opcua\lib\client\client_base.js:68
        throw new Error(" cannot locate certificate file "+options.certificateFile );
        ^

Error:  cannot locate certificate file node_modules\node-opcua\certificates\client_selfsigned_cert_102
4.pem`

erossignon, 2016-10-23:
the certificate file doesn't seems to exists  node_modules\node-opcua\certificates\client_selfsigned_cert_1024.pem

This could be due to the fact that the default certifcate creation has failed:
   - running behing a proxy ( preventing the openssl.exe utility to be downladed)
   - using an old version of openssl on MacOS;

You can run:
```
node node_modules\node-opcua\bin\crypto_create_CA.js demo
```
to create a new certificate store with sample certificates.

And make sure your client uses one of them, with its private key:

```javascript
var options = {
   certificateFile : "./certificates/client_selfsigned_cert_2048.pem"
   privateKeyFile: "./certificates/client_key_2048.pem"
};
var client = new opcua.OPCUAClient(options);
 ````

sebastianteres, 2016-11-02:
I'm getting the following error on mac while running crypto_create_CA.js demo:
start date is invalid, it should be YYMMDDHHMMSSZ
The command dates look like this: -startdate 20161101161257Z -enddate 20171101161257Z

mmautomatizacion, 2016-11-03:
Hi,
I'm having the same error, but nothing seems to work. I'm on a Mac with MacOS El Capitan. I've tried to update OpenSSL with brew, but is not posible to replace the OpenSSL of MacOS by the updated OpenSSL, I don't know why. I try to execute "crypto_create_CA.js", but error appear (the same of @sebastianteres). What can I do?

Thanks

sebastianteres, 2016-11-03:
@mmautomatizacion I found a quick solution (temporary).
On your node_modules folder find node-opcua-pki/lib/pki/toolbox.js
There you will find the function that generates the dates:  **x509Date**

Change the following line:

```js
return w(Y, 4) + w(M, 2) + w(D, 2) + w(h, 2) + w(m, 2) + w(s, 2) + "Z";
```

To this:

```js
return w(Y, 2) + w(M, 2) + w(D, 2) + w(h, 2) + w(m, 2) + w(s, 2) + "Z";
```

That will fix the dates and work correctly on Mac.
Hope it helps! :)

erossignon, 2016-11-03:
@sebastianteres @mmautomatizacion  the issue related to the old version of openssl on MacOS is discussed and solved here : #235

mmautomatizacion, 2016-11-04:
Thanks @erossignon, but I can't update OpenSSL with Brew, I don't know why. Maybe the solution is not for MacOS El Capitan.

mmautomatizacion, 2016-11-04:
Thanks @sebastianteres!!! The temporary solution works perfectly!!

erossignon, 2016-11-04:
@mmautomatizacion  have a look at http://stackoverflow.com/questions/35129977/how-to-install-latest-version-of-openssl-mac-os-x-el-capitan

mmautomatizacion, 2016-11-07:
Thanks @erossignon, but the user who wrote the post ensures that any answer has worked for him. For now, I think Node OPC-UA is not compatible with MacOS. At least, El Capitan.

erossignon, 2016-11-11:
@mmautomatizacion , using the command ```brew link openssl --force```  as explained in http://stackoverflow.com/a/35743303/406458 worked for me. NodeOPCUA is working fine on MacOS ,including El Captain, providing that you have a recent version of openssl.

closed 2016-11-11 by erossignon

PiotrDeyna, 2016-12-01:
I've installed MacOS 10.12.1 (Sierra) Xcode 8.1 and "brew install openssl".
Installed sample_server according instructions.
When I run sample_server:
sample_server.js

/Users/pd/node_modules/node-opcua/node_modules/better-assert/index.js:37
  throw err;
  ^
AssertionError: fs.existsSync(this.certificateFile), "Certificate file must exist :" + this.certificateFile
    at OPCUAServer.OPCUASecureObject.getCertificateChain (/Users/pd/node_modules/node-opcua/lib/misc/opcua_secure_object.js:61:9)
    at OPCUAServer.OPCUASecureObject.getCertificate (/Users/pd/node_modules/node-opcua/lib/misc/opcua_secure_object.js:48:33)
    at new OPCUAServer (/Users/pd/node_modules/node-opcua/lib/server/opcua_server.js:249:27)
    at Object.<anonymous> (/Users/pd/myserver/sample_server.js:3:14)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)

When I try to create certificate file with command 
"node bin/crypto_create_CA.js  certificate" 

....

    - creating the self signed certificate
    -------------------------------------- 

          set OPENSSL_CONF=/Users/pd/myserver/certificates/CA/conf/caconfig.cnf
                  OPENSSL_CONF /Users/pd/myserver/certificates/CA/conf/caconfig.cnf
                  RANDFILE     /Users/pd/myserver/certificates/random.rnd
                  CMD         openssl  ca  -selfsign  -keyfile "/Users/pd/myserver/certificates/client_key_1024.pem" -startdate 20161130085552Z -enddate 20171130085552Z -batch -out "/Users/pd/myserver/certificates/client_selfsigned_cert_1024.pem" -in "/Users/pd/myserver/certificates/client_selfsigned_cert_1024.pem_csr"
                  CWD          /Users/pd/myserver/certificates/CA
        stderr Using configuration from /Users/pd/myserver/certificates/CA/conf/caconfig.cnf
        stderr start date is invalid, it should be YYMMDDHHMMSSZ
        stderr 11962:error:0E06D06C:configuration file routines:NCONF_get_string:no value:/BuildRoot/Library/Caches/com.apple.xbs/Sources/OpenSSL098/OpenSSL098-64/src/crypto/conf/conf_lib.c:329:group=CA_default name=email_in_dn
 ########################################### OPENSSL ERROR  ###############################
Command failed: "openssl" ca  -selfsign  -keyfile "/Users/pd/myserver/certificates/client_key_1024.pem" -startdate 20161130085552Z -enddate 20171130085552Z -batch -out "/Users/pd/myserver/certificates/client_selfsigned_cert_1024.pem" -in "/Users/pd/myserver/certificates/client_selfsigned_cert_1024.pem_csr"
Using configuration from /Users/pd/myserver/certificates/CA/conf/caconfig.cnf
start date is invalid, it should be YYMMDDHHMMSSZ
11962:error:0E06D06C:configuration file routines:NCONF_get_string:no value:/BuildRoot/Library/Caches/com.apple.xbs/Sources/OpenSSL098/OpenSSL098-64/src/crypto/conf/conf_lib.c:329:group=CA_default name=email_in_dn

 ########################################### OPENSSL ERROR  ###############################
ERROR  Command failed: "openssl" ca  -selfsign  -keyfile "/Users/pd/myserver/certificates/client_key_1024.pem" -startdate 20161130085552Z -enddate 20171130085552Z -batch -out "/Users/pd/myserver/certificates/client_selfsigned_cert_1024.pem" -in "/Users/pd/myserver/certificates/client_selfsigned_cert_1024.pem_csr"
Using configuration from /Users/pd/myserver/certificates/CA/conf/caconfig.cnf
start date is invalid, it should be YYMMDDHHMMSSZ
11962:error:0E06D06C:configuration file routines:NCONF_get_string:no value:/BuildRoot/Library/Caches/com.apple.xbs/Sources/OpenSSL098/OpenSSL098-64/src/crypto/conf/conf_lib.c:329:group=CA_default name=email_in_dn

Command failed: "openssl" ca  -selfsign  -keyfile "/Users/pd/myserver/certificates/client_key_1024.pem" -startdate 20161130085552Z -enddate 20171130085552Z -batch -out "/Users/pd/myserver/certificates/client_selfsigned_cert_1024.pem" -in "/Users/pd/myserver/certificates/client_selfsigned_cert_1024.pem_csr"
Using configuration from /Users/pd/myserver/certificates/CA/conf/caconfig.cnf
start date is invalid, it should be YYMMDDHHMMSSZ
11962:error:0E06D06C:configuration file routines:NCONF_get_string:no value:/BuildRoot/Library/Caches/com.apple.xbs/Sources/OpenSSL098/OpenSSL098-64/src/crypto/conf/conf_lib.c:329:group=CA_default name=email_in_dn

....

what's wrong?

mmautomatizacion, 2016-12-01:
Use the command `brew link openssl --force`. For check, run the command `openssl version`. The OpenSSL version must be 1.0.0 or above. If not, restart the system and try again. In my case, It doesn't work for the first time.

PiotrDeyna, 2016-12-01:
Hi Luis,

Thank you for the quick answer!

I’ve tried 3 times to restart the computer and give the command as you’ve suggested.
It doesn’t work for me …

Piotrs-iMac:~ pd$ brew link openssl --force
Warning: Refusing to link: openssl
Linking keg-only openssl means you may end up linking against the insecure,
deprecated system OpenSSL while using the headers from Homebrew's openssl.
Instead, pass the full include/library paths to your compiler e.g.:
  -I/usr/local/opt/openssl/include -L/usr/local/opt/openssl/lib

Best regards, Piotr Deyna

PiotrDeyna, 2016-12-01:
Is not working for me; tried 3 times with restart.

pd$ brew link openssl --force
Warning: Refusing to link: openssl
Linking keg-only openssl means you may end up linking against the insecure,
deprecated system OpenSSL while using the headers from Homebrew's openssl.
Instead, pass the full include/library paths to your compiler e.g.:
  -I/usr/local/opt/openssl/include -L/usr/local/opt/openssl/lib

mmautomatizacion, 2016-12-01:
Ok. Try this:

`ln -s /usr/local/Cellar/openssl/1.0.2j/bin/openssl /usr/local/bin/openssl`
`brew link openssl --force`

Open another terminal an try `openssl version`

PiotrDeyna, 2016-12-01:
Thanks Luis!
I could replace OpenSSL version: 
pd$ openssl version
OpenSSL 1.0.2j  26 Sep 2016

I guess I could create certificate, but the sample_server.js terminates after start:

pd$ node bin/crypto_create_CA.js certificate

 using  config file  /Users/pd/myserver/node-opcua/certificates/config.js
 configuration = 
   commonName                     : NodeOPCUA-COMMONNAME
   organization                   : NodeOPCUA
   organizationUnit               : Unit
   locality                       : Paris
   state                          : IDF
   country                        : FR
   validity                       : 5475
   keySize                        : 2048
   certificateDir                 : /Users/pd/myserver/node-opcua/certificates
   CAFolder                       : /Users/pd/myserver/node-opcua/certificates/CA
   PKIFolder                      : /Users/pd/myserver/node-opcua/certificates/PKI
   privateKey                     : /Users/pd/myserver/node-opcua/certificates/PKI
   applicationUri                 : urn:Piotrs-iMac:Node-OPCUA-Server
          set ALTNAME_URI=urn:Piotrs-iMac:Node-OPCUA-Server
          set ALTNAME_DNS=Piotrs-iMac
          set ALTNAME_DNS_1=
          set ALTNAME_DNS_2=
          set ALTNAME_DNS_3=
          set ALTNAME_DNS_4=
          set ALTNAME_IP=

    - Creating a Certificate Signing Request
    ---------------------------------------- 

          set OPENSSL_CONF=/Users/pd/myserver/node-opcua/certificates/PKI/own/openssl.cnf
                  OPENSSL_CONF /Users/pd/myserver/node-opcua/certificates/PKI/own/openssl.cnf
                  RANDFILE     /Users/pd/myserver/node-opcua/certificates/random.rnd
                  CMD         openssl  req -new  -sha256  -batch  -text  -config "/Users/pd/myserver/node-opcua/certificates/PKI/own/openssl.cnf" -key "/Users/pd/myserver/node-opcua/certificates/PKI/own/private/private_key.pem" -out "/Users/pd/myserver/node-opcua/certificates/PKI/own/certs/certificate_1480596628628.csr"
          set OPENSSL_CONF=/Users/pd/myserver/node-opcua/node_modules/node-opcua-pki/lib/pki/empty_config.cnf
                  OPENSSL_CONF /Users/pd/myserver/node-opcua/node_modules/node-opcua-pki/lib/pki/empty_config.cnf
                  RANDFILE     /Users/pd/myserver/node-opcua/certificates/random.rnd
                  CMD         openssl  version
                  CWD          .
        stdout OpenSSL 1.0.2j  26 Sep 2016
                  CWD          /Users/pd/myserver/node-opcua/certificates/PKI
 csr_file =  /Users/pd/myserver/node-opcua/certificates/PKI/own/certs/certificate_1480596628628.csr
 start Date  Thu, 01 Dec 2016 12:50:28 GMT 20161201125028Z
 end   Date  Fri, 01 Dec 2017 12:50:28 GMT 20171201125028Z
          set ALTNAME_URI=urn:Piotrs-iMac:Node-OPCUA-Server
          set ALTNAME_DNS=Piotrs-iMac
          set ALTNAME_DNS_1=
          set ALTNAME_DNS_2=
          set ALTNAME_DNS_3=
          set ALTNAME_DNS_4=
          set ALTNAME_IP=

    - then we ask the authority to sign the certificate signing request
    ------------------------------------------------------------------- 

          set OPENSSL_CONF=/Users/pd/myserver/node-opcua/node_modules/node-opcua-pki/lib/pki/empty_config.cnf
                  OPENSSL_CONF /Users/pd/myserver/node-opcua/node_modules/node-opcua-pki/lib/pki/empty_config.cnf
                  RANDFILE     /Users/pd/myserver/node-opcua/certificates/random.rnd
                  CMD         openssl  ca  -config conf/caconfig.cnf  -startdate 20161201125028Z -enddate 20171201125028Z -batch -out "/Users/pd/myserver/node-opcua/certificates/PKI/own/certs/certificate_1480596628628.pem" -in "/Users/pd/myserver/node-opcua/certificates/PKI/own/certs/certificate_1480596628628.csr"
                  CWD          /Users/pd/myserver/node-opcua/certificates/CA
        stderr Using configuration from conf/caconfig.cnf
        stderr Check that the request matches the signature
        stderr Signature ok
        stderr The Subject's Distinguished Name is as follows
        stderr countryName           :PRINTABLE:'FR'
[... 38 lines cut ...]
    ----------------------------------- 

    - construct certificate chain
    ----------------------------- 

    - verify certificate against the root CA

**************************

pd$ node sample_server.js

/Users/pd/node_modules/node-opcua/node_modules/better-assert/index.js:37
  throw err;
  ^
AssertionError: fs.existsSync(this.certificateFile), "Certificate file must exist :" + this.certificateFile
    at OPCUAServer.OPCUASecureObject.getCertificateChain (/Users/pd/node_modules/node-opcua/lib/misc/opcua_secure_object.js:61:9)
    at OPCUAServer.OPCUASecureObject.getCertificate (/Users/pd/node_modules/node-opcua/lib/misc/opcua_secure_object.js:48:33)
    at new OPCUAServer (/Users/pd/node_modules/node-opcua/lib/server/opcua_server.js:249:27)
    at Object.<anonymous> (/Users/pd/myserver/sample_server.js:3:14)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)

PiotrDeyna, 2016-12-01:
Hi Luis!
Thanks for your support.
It solved my problem.
I discovered that I've had two versions of node-opcua on my computer, one under my root catalogue/node_modules and another on myserver/node_modules.
I've removed both, installed one only and i works fine now.
Thanks!

Razzeee, 2017-11-09:
I ended up going into the node_modules folder for node_opcua and manually running `node .\postinstall.js`. Then figured out that while he downloads and tries to unzip openssl, he only does it for the first file and then exits. I then just unzipped the file by hand and did run `postinstall.js` again and it worked.

erossignon, 2017-11-09:
@Razzeee you must be using a very old version, this unzipping file issue has been solved recently, I suggest you update to the latest ( v0.1.0-12)

Razzeee, 2017-11-09:
@erossignon Correct, we're on v0.0.65 mostly because it looked like it was the latest release due to https://github.com/node-opcua/node-opcua/releases and no changelog elsewhere.
Thanks for getting back.

limengsong, 2019-08-30:
@erossignon 
Hello, I don’t know why it doesn’t load.
```
const opcua = require("node-opcua");
const async = require("async");
const endpointUrl = 'opc.tcp://172.26.15.7:48010';
var SecurityPolicy = opcua.SecurityPolicy;
var MessageSecurityMode = opcua.MessageSecurityMode;

const client = opcua.OPCUAClient.create({
    securityMode: MessageSecurityMode.SIGNANDENCRYPT,
    securityPolicy: SecurityPolicy.Basic128Rsa15,
    requestedSessionTimeout: 10000,
    applicationName: "NodeOPCUA-Client",
    endpoint_must_exist: false,
    certificateFile: "./own/certs/uaclientcpp.der",
    privateKeyFile: "./own/private/uaclientcpp.pem",
});
```

```
PS D:\HolliEBR--Demo\node-opcua-samples--0829> node .\bin\test_scada.js
D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\node-opcua-client\dist\private\client_base_impl.js:195
            throw new Error(" cannot locate certificate file " + options.certificateFile);
            ^

Error:  cannot locate certificate file ./own/certs/uaclientcpp.der
    at new ClientBaseImpl (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\node-opcua-client\dist\private\client_base_impl.js:195:19)
    at new OPCUAClientImpl (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\node-opcua-client\dist\private\opcua_client_impl.js:239:9)
    at Function.create (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\node-opcua-client\dist\opcua_client.js:9:16)
    at Object.<anonymous> (D:\HolliEBR--Demo\node-opcua-samples--0829\bin\test_scada.js:16:34)
    at Module._compile (internal/modules/cjs/loader.js:688:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:699:10)
    at Module.load (internal/modules/cjs/loader.js:598:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:537:12)
    at Function.Module._load (internal/modules/cjs/loader.js:529:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:741:12)
PS D:\HolliEBR--Demo\node-opcua-samples--0829>
```

erossignon, 2019-08-30:
as said in the error message, node-opcua cannot locate the certificate file
`Error:  cannot locate certificate file ./own/certs/uaclientcpp.der`
May be you need to use absolute path rather than  relative.

limengsong, 2019-08-30:
@erossignon 
Thank you！
The path is changed, the connection is successful, but when you createSession, you will get an error.
My Codes:
```
const opcua = require("node-opcua");
const async = require("async");
//var crypto_utils = require("lib/misc/crypto_utils");
const endpointUrl = 'opc.tcp://172.26.15.7:48010';
var SecurityPolicy = opcua.SecurityPolicy;
var MessageSecurityMode = opcua.MessageSecurityMode;

const client = opcua.OPCUAClient.create({
    securityMode: MessageSecurityMode.SIGNANDENCRYPT,
    securityPolicy: SecurityPolicy.Basic128Rsa15,
    requestedSessionTimeout: 10000,
    applicationName: "NodeOPCUA-Client",
    endpoint_must_exist: false,
    certificateFile: "certificates/PKI/own/certs/uaclientcpp.der",
    privateKeyFile: "certificates/PKI/own/private/uaclientcpp.pem",
  //  serverCertificate: crypto_utils.readCertificate("trusted_certificates/cert.pem")
});
```

Error Informations:
```
PS D:\HolliEBR--Demo\node-opcua-samples--0829> node .\bin\test_scada.js
connected !
 failure  Error:  End point must exist opc.tcp://172.26.15.7:48010
    at OPCUAClientImpl._createSession (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\node-opcua-client\dist\private\opcua_client_impl.js:777:29)
    at OPCUAClientImpl.createSession (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\node-opcua-client\dist\private\opcua_client_impl.js:264:14)
    at OPCUAClientImpl.createSession (eval at thenify.withCallback (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\thenify\index.js:33:10), <anonymous>:6:48)
    at OPCUAClientImpl.createSession (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\node-opcua-client\dist\private\opcua_client_impl.js:258:25)
    at OPCUAClientImpl.createSession (eval at thenify.withCallback (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\thenify\index.js:33:10), <anonymous>:6:48)
    at D:\HolliEBR--Demo\node-opcua-samples--0829\bin\test_scada.js:49:16
    at eachfn (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\async\dist\async.js:2952:28)
    at replenish (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\async\dist\async.js:444:21)
    at iterateeCallback (D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\async\dist\async.js:428:21)
    at D:\HolliEBR--Demo\node-opcua-samples--0829\node_modules\async\dist\async.js:325:20
```