node-opcua/node-opcua #257: How read and write of 10k+ OPC UA nodes

biancode, 2016-10-07:
Dear Etienne,

I'm testing with large information models 10k+ and it becomes very slow.
I'm searching for the OPC UA node in address space and if it doesn't exist it setups the node, otherwise it writes the value to the node. 
**Please, can you explain a bit how you would work with 10k+ nodes?**

Example: 

- 10 folders with á 10 subfolders including 100 variables
- 20 folders with á 10 subfolders including 100 variables

Until 10k it works but from 10k+ it becomes very slow and it looks like a exponential problem in searching a big address space.

searching is like: serverAddressSpace.findNode("ns=0;i=85")

kind regards
Klaus

biancode, 2016-10-07:
maybe BaseNode.prototype.findReferences

mikakaraila, 2016-10-17:
I think continuation point is not yet supported...

erossignon, 2016-10-27:
@biancode Klaus,
please provide an example of the code you're using to "find" for a node , so we see exactly how you're trying to use the library.
Are you trying to find the node by nodeID ? By browseName ? By path  from the root folder ?  By looking at some references ? 

BaseNode#findReferences is meant to find all references attached to a node with a given reference type. There is usually many. A simple dictionary cannot be used in this case.

closed 2017-05-19 by biancode