node-opcua/node-opcua #258: Dynamic Variables

gremlinc5, 2016-10-08:
Hi, I got a problem. I use opcua server to sync with my custom server via TCP Socket. It do well. But when i write from a opc client to opc ua it write all the variables. I think it's the variable initialization the problem.

`
	function construct_my_address_space(server, tags) {
		var addressSpace = server.engine.addressSpace;

		// declare a new object
		var device = addressSpace.addObject({
			organizedBy: addressSpace.rootFolder.objects,
			browseName: "AntServer"
		});
		
		console.log(tags.Type);
		
		var variables = [];
		
		for (var i = 0; i < tags.GetTags.length; i++)
		{
			variables.push(0);
			
			addressSpace.addVariable({
				componentOf: device,
				browseName: tags.GetTags[i].Tag,
				dataType: "Double",
				value: {
					get: function () {
						return new opcua.Variant({dataType: opcua.DataType.Double, value: variables[i] });
					},
					set: function (variant) {
						variables[i] = parseFloat(variant.value);
						return opcua.StatusCodes.Good;
					}
				}
			});
			
			console.log(tags.GetTags[i].Tag);
		}
		
		var os = require("os");
		/**
		 * returns the percentage of free memory on the running machine
		 * @return {double}
		 */
		function available_memory() {
			// var value = process.memoryUsage().heapUsed / 1000000;
			var percentageMemUsed = os.freemem() / os.totalmem() * 100.0;
			return percentageMemUsed;
		}
		
		addressSpace.addVariable({

			componentOf: device,

			nodeId: "ns=1;s=free_memory", // a string nodeID
			browseName: "FreeMemory",
			dataType: "Double",    
			value: {
				get: function () {return new opcua.Variant({dataType: opcua.DataType.Double, value: available_memory() });}
			}
		});
	}
`

I'm using code like the example but how could I use dynamic variables with it?

Thank you

gremlinc5, 2016-10-09:
I found this solution:

`var variables = {};
		
		for (var i = 0; i < tags.GetTags.length; i++)
		{
			variables[tags.GetTags[i].Tag] = {"value" : 0.0, "is_set" : false};
			
			addressSpace.addVariable({
				componentOf: device,
				browseName: tags.GetTags[i].Tag,
				dataType: "Double",
				value: {
					get: function () {
						return new opcua.Variant({dataType: opcua.DataType.Double, value: variables[this["browseName"]] });
					},
					set: function (variant) {
						variables[this["browseName"]].value = parseFloat(variant.value);
						variables[this["browseName"]].is_set = true;
						return opcua.StatusCodes.Good;
					}
				}
			});
			
			console.log(tags.GetTags[i].Tag);
		}`

closed 2016-10-09 by gremlinc5

erossignon, 2016-10-27:
this is related to understanding how javascript "closures" work.
( see http://stackoverflow.com/questions/111102/how-do-javascript-closures-work )