node-opcua/node-opcua #259: address space problems with multiple servers

biancode, 2016-10-09:
Dear Etienne,

I tried to use multiple OPC UA servers and I got some problems with the address space.
Just one server of n servers has a valid reachable address space. All other servers didn't find any added reference or node in address space.

like:
`
server_one = new opcua.OPCUAServer({
                port: 53880,
                nodeset_filename: xmlFiles,
                resourcePath: "UA/NodeRED/ISA95Server",
                buildInfo: {
                    productName: node.name.concat("OPC UA server"),
                    buildNumber: "1911",
                    buildDate: new Date(2016, 9, 25)
                }
            });
`

`
server_two = new opcua.OPCUAServer({
                port: 53881,
                nodeset_filename: xmlFiles,
                resourcePath: "UA/NodeRED/ISA95Server",
                buildInfo: {
                    productName: node.name.concat("OPC UA server"),
                    buildNumber: "1911",
                    buildDate: new Date(2016, 9, 25)
                }
            });
`

Do you have an idea what is wrong in that case?

biancode, 2016-10-09:
The server init takes 8 minutes for 6 OPC UA servers. I mean there are some arrays or one that brings it to an exponential amount to init the servers.

all on RPI2: If I use one server it takes **22** sec to init and six servers brings every server to **84** sec in sum **504** sec (new opcua.OPCUAServer) to init six servers at the same environment. Address space goes from one with **1,2** sec  to six with **5,8** sec (serverAddressSpace.addObject) with a grow from 250 OPC UA nodes/items to 20030. That is much faster than the init is.

erossignon, 2016-10-27:
I have created a gist to try reproduce this issue: https://gist.github.com/erossignon/b188a0f84254034e6c92df0e82d1a8ac

In this gist, I am creating 2 servers and for each server I am creating a new variable. Checking with UAExpert I have the expected behavior server1.variable1 is only visible on server1 whereas server2.variable2 is only visible on server2.

erossignon, 2017-04-06:
@biancode  any progress on this ? shall we close ?

biancode, 2017-04-07:
Yes, it World for me with 0.0.61.

closed 2017-04-07 by erossignon