node-opcua/node-opcua #263: How to run node-opcua on mobile devices?

jdiegogonzales, 2016-10-17:
Has anyone tried to ship a node-opcua web client via Apache Cordova, JXcore, Termux or something similar? Help would be very much appreciated.

Thanks

BLamy, 2016-10-25:
You will need to run node-opcua in a node environment. Hybrids like electron will also work. However, mobile platforms only allow execution of javascript from a browser context. You may be able to try using JXCore but I have my doubts.

erossignon, 2017-01-01:
See this [wiki page ](https://github.com/node-opcua/node-opcua/wiki/installing-node-opcua-on-your-Android-Phone-with-Termux) that explains how to run [opcua-commander](https://github.com/node-opcua/opcua-commander) from a Android device using Termux.
Let me know if this help and please complement the wiki page with your findings

closed 2018-01-31 by erossignon