node-opcua/node-opcua #264: How to connect to an OPC Server?

echaparroampa, 2016-10-20:
Hi there, I've recently discovered this tool, and I'd like to know if I could connect to a regular OPC Server using "Node-OPCUA" (A server such as "MatrikonOPC", "Top Server", etc )

erossignon, 2016-10-23:
simply said : you can't.

node-opcua uses the "Unified Automation".
OPC classic is a 20 year old standard relying on OLE or DCOM technology on Microsoft Windows.
OPCUA is the a new and modern standard based on TCP and operating on all type of devices.
OPC  is being gradually replaced with OPCUA 
see http://stackoverflow.com/questions/30642918/difference-between-opc-ua-and-opc-da

node-opcua focuses exclusively on OPC-UA.

jayantad, 2016-10-27:
Well, you can add a Wrapper (DA/UA) over your classic OPC DA servers to create a UA server. Then you can connect to this UA server using any UA client.

see https://www.matrikonopc.com/opc-ua/products/ua-wrapper-com-opc-servers.aspx

mininggold, 2016-11-01:
You could use Kepware as a transparent link.
See the following document:
on page 4 under section 5.1.3 point 18 and the Tip mentions: 
"Without importing tags, users can address OPC items in the target OPC DA server across the KEPServerEX OPC tunnel by using a particular tag address syntax from the client application connected to the tunnel client"
example:
MyOPCUAchannel1.myOPCUAdevice1.ns=2;s=myOPCDAchannel2.myOPCDAdevice2.tag

closed 2017-01-04 by erossignon