node-opcua/node-opcua #266: Websocket problems

antmag, 2016-11-01:
Hello,

Im trying to read opcua variables with socket.io and monitoreditem.on(change) 
I can read the values but I have got an issue! When I'm on the chrome websocket console I can show my datas in real time but each 25 seconds I have got two datas ( 2 and 3). If I don't receive any value between this new datas each 25 secondes i can not anymore read the values ...

antmag, 2016-11-01:
[image]
[image]
[image]
[image]

antmag, 2016-11-01:
I think I need to change the ClientSubscription settings but it doesnt work..
The third picture is the picture of the websocket console when the connection is turn off because no values changes between the first data 2 3 and the second

antmag, 2016-11-04:
I fixed the problem

erossignon, 2016-12-06:
@antmag You reopened the issue. Have you fixed the problem ?

antmag, 2016-12-06:
I found an other way to so but I didnt solved this issue

erossignon, 2016-12-06:
I would use a smaller maxKeepAliveCount ( such as 60) just to make sure that the client and the server exchange some data every maxKeepAliveCount*publishingInterval milliseconds , this will prevent the connection and session to timeout.

closed 2018-01-31 by erossignon