node-opcua/node-opcua #267: Sending server certificate

constantinbeleca, 2016-11-01:
Upon connecting to a node-opcua server pretty much the same as the one presented in the 'simple_server' example, from the windwos client from Integration Objects I get the following error and can't connect: 

"Session creation failed. Exception: Server did not return the certificate used to create the secure channel."

From UaExpert, the client from Inductive Automation I can connect to the server but I always get the following message:

"BadCertificateChainIncomplete" with the errors "unable to get local issuer certificate[BadCertificateChainIncomplete]", "unable to get certificate CRL[BadCertificateRevocationUnknown]" and "unable to verify the first certificate[BadCertificateChainIncomplete]". If I select "Accept the server certificate temporarily for this session" I can connect to my node-opcua server but I would pretty much like to not have to do this every time.

I've searched between older posts but couldn't find anything helpful.
Am I missing something?

erossignon, 2017-01-04:
node-opcua now  use self-signed certificate by default (commit 3a2092235311235ac810d2ca204d094016b38aa2 )

erossignon, 2019-04-25:
Issuer certificate must be trusted on client side ...

closed 2019-04-25 by erossignon