node-opcua/node-opcua #268: Client side dynamic variables to add/remove on the fly

mininggold, 2016-11-01:
Relates to this issue:
[https://github.com/node-opcua/node-opcua/issues/250](url)

What is the best method for a client to read multiple variables at the same time, and change the list of variables dynamically on the fly. 
This is for use with a single Node server, similar to:
[https://github.com/node-opcua/node-opcua-htmlpanel](url)
except there are multiple html pages each with a different list of variables, coming from the same OPC UA server and the same Node HTTP web server.
The server side code does not have a static list of OPC variables, since when the html page is opened the variable list should be added dynamically to the server side and then removed from either an OPC UA polling or subscription list when the html page is closed (not being viewed). This is to reduce the load on the OPC UA server which supplied by a vendor such as Emerson DeltaV.

erossignon, 2016-12-06:
* the ```browse service``` will allow you to discover  nodes inside a server 

Check a example here https://github.com/node-opcua/node-opcua/blob/master/bin/simple_client.js#L268

mininggold, 2016-12-06:
Thank you; the browser service is one option, though for this application all tags are known ahead of time, so browsing will not be used, since tags will be stored in a database and searched from there. 

The tag list for this application will change dynamically which means adding and removing tags from a subscription on the fly.
Option: [https://github.com/coussej/node-opcua-logger](https://github.com/coussej/node-opcua-logger)

This comes close: [http://stackoverflow.com/questions/34373347/node-js-opc-ua-many-monitored-items](http://stackoverflow.com/questions/34373347/node-js-opc-ua-many-monitored-items)

Question: does the connection/subscription need to be closed 1st before adding/removing tags from the monitored items of the subscription?

erossignon, 2016-12-06:
Non, monitored items can be added or deleted from a subscription dynamically at runtime. There is no need to recreate a session.

closed 2017-01-04 by erossignon

fwaznad, 2018-09-03:
hi @erossignon, how to delete monitored Items at runtime?