node-opcua/node-opcua #271: How to handle timeouts

lars-magne, 2016-11-12:
In my client code (based on the simple_client) I try to connect to a non-existing endpoint hope that the code should enter the function with "err" and write "cannot connect to endpoint" in the console.  I also struggle to reconnect a subscription if I loose the connection and want it to automatically reconnect and reestablish monitored items when the connection is back, .

Any idea how to solve this?

Regards
Lars Magne Endresen

erossignon, 2016-11-18:
By default client will retry to connect continuously to the server, if the server is not available.
This gives you the possibility to start the client before you start the server for instance.

You can control the retry/bakoff strategy by setting the ```connectionStrategy.maxRetry``` options:

```javascript
 var options = {
    connectionStrategy: {
        maxRetry: 1
    }
  };
  var client = new opcua.OPCUAClient(options);
```

You can also set some interesting event handler that will let you know what's going on during the connect phase:

```javascript
     client.on("start_reconnection", function() {
        console.log(" ... start_reconnection")
      });
      client.on("backoff", function(nb, delay) {
        console.log("  connection failed for the", nb,
          " time ... We will retry in ", delay, " ms");
      });
      client.connect(endpointUrl, function(err) {
        if (err) {
          console.log(" cannot connect to endpoint :", endpointUrl);
        } else {
          if (doDebug) {
            console.log("connected !");
          }
        }
        callback(err);
      });
````

KENZoft, 2017-01-09:
When trying to reconnect to a CoDeSys OPC-UA server I get  **BadSessionIdInvalid (0x80250000)** error and then the client stops trying to reconnect.

Complete response:

err =  { [Error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2017-01-09T06:49:20.582Z
   requestHandle               /* IntegerId                        */: 2               0x2
   serviceResult               /* StatusCode                       */: BadSessionIdInvalid (0x80250000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};]
  response:
   ServiceFault {
     responseHeader:
      ResponseHeader {
        timeStamp: Mon Jan 09 2017 07:49:20 GMT+0100 (Västeuropa, normaltid),
        requestHandle: 2,
        serviceResult: [Object],
        serviceDiagnostics: [Object],
        stringTable: [Object],
        additionalHeader: null } } }
warning : disconnection : closing pending sessions

Any idea?

mmautomatizacion, 2017-02-27:
I have the last version, 0.0.61. I'm using Node-OPCUA in client mode for read some data cyclically from OPC-UA server with "ClientSubscription". It starts correctly, but when I shutdown server, It only stop read data. No error returned. When server start again, nothing happen. I have to restart client for obtain data again. 

I have some events but no one is fired. This is the code for connect:
```
self.client.on('start_reconnection', function() {
	console.log("START RECONNECTION");
});
self.client.on('after_reconnection', function() {
	console.log("AFTER RECONNECTION");
});
self.client.on('backoff', function(nb, delay) {
	console.log("  connection failed for the", nb, " time ... We will retry in ", delay, " ms");
});
self.client.connect(self.endpointUrl, function (err) {
    if(err) {
        console.log("Unable to connect to" , self.endpointUrl );
    }
});
```

And this the code for subscription:
```
var subscription = new opcua.ClientSubscription(clientSession, {
    requestedPublishingInterval: 1000,
    requestedLifetimeCount: 1000,
    requestedMaxKeepAliveCount: 2,
    maxNotificationsPerPublish: 1000,
    publishingEnabled: true,
    priority: 1
});

subscription.on('started', function() {
	console.log("Subscription started [" + subscription.subscriptionId + "]");
});

var value = subscription.monitor({
	nodeId: nodeId,
	attributeId: opcua.AttributeIds.Value
},
{
	samplingInterval: 1000,
	discardOldest: true,
	queueSize: 100
});
```

What Am I doing wrong?

FRules, 2018-02-07:
Hi,
is it possible to decrease the timeout? I already tried to adjust the connectionStrategy parameters. The client initialization code looks like this now. 
```js
var client = new opcua.OPCUAClient({
    connectionStrategy: {
        maxRetry: 0,
        initialDelay: 100,
        maxDelay: 200
    }
});
```
Unfortunately, at least for me, those settings had no effect. When I set maxRetry to 100, the output on the console was as expected, but instead of the 100/200 ms reconnection delay it still took 20 seconds. 

```
connection failed for the 0  time ... We will retry in  100  ms
connection failed for the 1  time ... We will retry in  200  ms
connection failed for the 2  time ... We will retry in  200  ms
connection failed for the 3  time ... We will retry in  200  ms
```

erossignon, 2018-02-07:
* maxRetry: the number of retries before the reconnection process will give up and return a connection  error:
* initialDelay: the delay in milisecond before the first reconnection occur ( this delay will increase linearly if reconnection is still failing ( until it reaches maxDelay)
* maxDelay: the maximum delay between two attempts to reconnect

FRules, 2018-02-08:
Maybe I did something wrong but changing the parameters does not work for me. I added a timestamp to my log. 

```
2018-02-08 08:05:53   connection failed for the 0  time ... We will retry in  100  ms
2018-02-08 08:06:14   connection failed for the 1  time ... We will retry in  200  ms
2018-02-08 08:06:35   connection failed for the 2  time ... We will retry in  200  ms
2018-02-08 08:06:56   connection failed for the 3  time ... We will retry in  200  ms
2018-02-08 08:07:17   connection failed for the 4  time ... We will retry in  200  ms
```

Shouldn't it be the case that the reconnections should occur much faster? Or does the connection attempt itself last 20 seconds?

erossignon, 2019-03-26:
reconnection has been improved since this issue was raised. I would be inclined to close this issue

closed 2019-03-26 by erossignon

JesseJ12345, 2019-04-30:
@KENZoft , Did you fix this problem with   **BadSessionIdInvalid (0x80250000)** ?  Did you have any suggestion ? I also meet this bug

erossignon, 2019-04-30:
This issue is closed. 
The problem BadSessionIdInvalid may deserve a  dedicated issue. I suggest you open one.
Do not forget to fill the template and provide a precise descriptions of the issue, step to reproduce, and sample source code that highlight the problem.