node-opcua/node-opcua #272: How to get DataType

mmautomatizacion, 2016-11-14:
Hi,
I'm very new in OPC-UA. I have to create a client that write an array in a server that each element contains a structure. In the node-opcua examples we have to define the DataType of each element and write it one by one. With UaExpert I can see how long is the array, the components of structures, even the DataType of them. How can I get that information with node-opcua? Can I write the structure directly from a javascript object?

Thanks

erossignon, 2016-12-06:
*  How can I get that information _(aka DataType )_ with node-opcua? 

```javascript
var node_to_read = {
   nodeId:  "ns=0;i=2253",
   attributeId: opcua.AttributeIds.DataType
};
client.read(node_to_read,function (err,dataValue) {
  ...
});
```

*Can I write the structure directly from a javascript object? 
   please , be more specific ...

mmautomatizacion, 2016-12-07:
Ok. I give you an example:

Imagine I have this structure in OPC-UA server:
```
 - Array
    [0]: Object
        - Parameter 1 (Int16)
        - Parameter 2 (Uint32)
        - Parameter 3 (String)
    [1] Object
        - Parameter 1 (String)
        - Parameter 2 (String)
    ...
```
I have created an array in javascript with the same objects and I have defined the values. I want to write all the values in OPC-UA server.

Actually, I have to define whats elements compounds the array, all the parameters and the type of each one for write the value. The problem is that probably I will change the structure in a future and I think is very tedious to redefine the object in OPC-UA server, redefine the javascript array and redefine the name and datatype of each element for write.

I'm searching a method for obtain all the elements of an array, or all the elements of a structure, and its datatype. Now, I'm using the "crawler" method for obtain the "nodeid" of all the elements of an structure, and with "forEach" function I generate a "nodesToRead" array for obtain the datatype of each one. Is there any method for do this with only one function? That is, a method for obtain all the tree of an array or structure and the datatype of each element, similar to crawler but obtaing the datatype too.

Thanks.

KENZoft, 2016-12-23:
Only works when using an array:

`                        var nodes_to_read = [{
                            nodeId: reference.nodeId,
                            attributeId: opcua.AttributeIds.DataType
                        }];
                        client.read(nodes_to_read, function(err, dataValue) {
                            console.log(dataValue);})`

closed 2018-01-25 by erossignon