node-opcua/node-opcua #273: Unable to write: "Not the same type"

mmautomatizacion, 2016-11-15:
Hi,
I'm having problem to write a variable to an OPCUA server. I've tried with booleans, integers, float... Always returns an 'BadTypeMismatch' error. This is the code I have using: 

```
var nodesToWrite = new Array();

nodesToWrite.push({
	nodeId: 'ns=6;s=::AsGlobalPV:test',
    	attributeId: opcua.AttributeIds.Value,
    	indexRange: null,
    	value: {
               	dataType: opcua.DataType.Boolean,
		value: true
	}
});

clienteSesion.write(nodesToWrite, function(err, status, diag) {
	console.log(err);
	console.log(status);
	console.log(diag);
});
```

The variable _err_ returns null and diag returns an empty array.
_Status_ returned this object:
```
[ { [Number: 2155085824]
    value: 2155085824,
    description: 'The value supplied for the attribute is not of the same type as the attribute\'s value.',
    name: 'BadTypeMismatch' } ]
```

What am I doing wrong?

Thank you!

erossignon, 2016-11-18:
You can use ```opcua_commander```  to navigate to the node and display its attributes, one of this will be the requested DataType.
https://github.com/node-opcua/opcua-commander

mmautomatizacion, 2016-11-18:
First I try to read the DataType directly from node-opcua. This is the function I had used for read the DataType:

```
clienteSesion.readVariableValue(nodeToRead, function(err, result) {
	var datatype = result[0].value.$dataType;
});
```

The variable _datatype_ was directly used like dataType parameter for write. The result is the same.

erossignon, 2016-11-18:
Correct, this is an other way of doing. I suggest that you use ```result[0].value.dataType;``` without the ```$``` sign.
You can use the ```toString``` method to get a pretty debug string 
```console.log(result[0].toString());

mmautomatizacion, 2016-11-18:
Thanks but, with the result, I'm trying to write into this value and It return 'BadTypeMismatch' error. Supposedly the type is correct, cause It have been obtained directly by the read function.

mmautomatizacion, 2016-11-18:
Here is the complete example:

```
session.readVariableValue('ns=6;s=::AsGlobalPV:test', function(err, result) {

	var nodesToWrite = new Array();
	
	nodesToWrite.push({
		nodeId: 'ns=6;s=::AsGlobalPV:test',
		attributeId: opcua.AttributeIds.Value,
		indexRange: null,
		value: {
			dataType: result.value.dataType,
			value: true
		}
	});

	session.write(nodesToWrite, function(err, status, diag) {
		console.log(err);
		console.log(status);
		console.log(diag);
	});
});
```

The example returns the error `TypeError: Cannot create property 'dataType' on boolean 'true'`. For resolve it I have used the solution on the post [#157](https://github.com/node-opcua/node-opcua/issues/157#issuecomment-174504198). The result is the '_BadTypeMismatch_' error I mentioned in the first message.

```
session.readVariableValue('ns=6;s=::AsGlobalPV:test', function(err, result) {

	var nodesToWrite = new Array();

	nodesToWrite.push({
		nodeId: 'ns=6;s=::AsGlobalPV:test',
		attributeId: opcua.AttributeIds.Value,
		indexRange: null,
		value: {
			value: {
				serverTimestamp: new Date(2015, 5, 2),
				serverPicoseconds: 20,
				sourceTimestamp: new Date(2015, 5, 3),
				sourcePicoseconds: 30,
				value: {
					dataType: result.value.dataType,
					value: true
				}
			}
		}
	});

	session.write(nodesToWrite, function(err, status, diag) {
		console.log(err);
		console.log(status);
		console.log(diag);
	});
});
```

Thanks!!

erossignon, 2016-11-19:
Correct , this is a duplicate of #157. closing then.

mmautomatizacion, 2016-11-19:
Please, Have you read something I have written? Even with the #157 solution I'm having the 'BadTypeMismatch' error. I've tried anything without success. The error is no closed.

erossignon, 2016-11-20:
method 1 using UAExpert or opcua_commander
  * navigate to ns=6;s=::AsGlobalPV:test'
  * read the DataType 

method 2:
  * use this gist
 <script src="https://gist.github.com/erossignon/8d80ae05379a4b4c5fcbf83458432e70.js"></script>

mmautomatizacion, 2016-11-21:
Thanks @erossignon,

The method 1 return me this: 
```
Datatype: Boolean
    NamespaceIndex: 0
    IdentifierType: Numeric
    Identifier: 1
```

I will try method 2.

mmautomatizacion, 2016-11-21:
Method 2 return me the same, type Boolean: `{ [Number: 1] key: 'Boolean', value: 1 }`

I have a questions. When I try to write, the status returns a number. What's this number? A error code maybe?
```
{ [Number: 2155085824]
    value: 2155085824,
    description: 'The value supplied for the attribute is not of the same type as the attribute\'s value.',
    name: 'BadTypeMismatch' }
```

Thanks.

mmautomatizacion, 2016-11-21:
I have checked that in "ClientSession.prototype.write" function, when "performMessageTransaction", the "request" variable has lost all information about value.

I researched and I found the value lost in "_auto_generated_DataValue.js" file, exactly in DataValue function, when It try to create a new Variant (line 58). The "options.value" variable contains correct information about value, but It return empty value structure to "self.value".

mmautomatizacion, 2016-11-21:
Ok, sorry. I have found an error in my code and I'm using a value structure into the original value structure on "nodesToWrite" array:

```
nodesToWrite.push({
        nodeId: 'ns=6;s=::AsGlobalPV:test',
        attributeId: opcua.AttributeIds.Value,
        indexRange: null,
        value: {
            value: {
                serverTimestamp: new Date(2015, 5, 2),
                serverPicoseconds: 20,
                sourceTimestamp: new Date(2015, 5, 3),
                sourcePicoseconds: 30,
                value: {
                    dataType: result.value.dataType,
                    value: true
                }
            }
        }
    });
```

Now, I'm getting  the error 'BadWriteNotSupported: The server not does support writing the combination of value'. What does combination of value mean? What serverTimestamp and sourceTimestamp must be used?

Thanks

erossignon, 2016-11-21:
Try to set  ```serverTimestamp: null, serverPicoseconds: 0```. Some server doesn't like Write with serverTimestamp

closed 2016-11-23 by erossignon