node-opcua/node-opcua #275: translateBrowsePath and KepServerEx6

gbaquedano, 2016-11-21:
Hello,
I had no trouble with doing this on KepServerEx5, but for some reason it is not working fine on version 6. Not sure if this is node-opcua issue though.
I'm attempting to recover node id's from browseNames, for that I created a simulator device with a string variable. This is what I get by browsing, which is correct:

* 2:Simulation [ns=2;s=Simulation]
* 2:Simulation.Device1 [ns=2;s=Simulation.Device1]
* 2:Simulation.Device1.Item1 (String) [ns=2;s=Simulation.Device1.Item1]

So I attempted to do the reverse operation:

`_opcua_session.translateBrowsePath(opcua.browse_service.makeBrowsePath("ObjectsFolder", "/2:Simulation.Device1.Item1"), function (err, results) {
                if (err) {
                    console.log("Err:" + err);
                } else {
                    console.log(results);
                }
            });`
which will return allways:

BrowsePathResult {
  statusCode:
   { [Number: 2154758144]
     value: 2154758144,
     description: 'The requested operation has no match to return.',
     name: 'BadNoMatch' },
  targets: [] }

Browsing to ("ObjectsFolder", "/2:Simulation") works though. Same happens browsing to "RootFolder/Objects/..." Any ideas?

erossignon, 2016-11-23:
Please check the namespace index of the BrowseName at each item of your path.
May be  "/2:Simulation.2:Device1.2:Item1" will do

gbaquedano, 2016-11-25:
Hi Etienne,
I tried but it wont work, I keep getting the "BadNoMatch" and I think I have tried all possible paths.

erossignon, 2016-11-26:
Is there any possibility for you to expose a sample server based on KepServerEx6 over the internet so we can remotely explore the issue ?

erossignon, 2016-11-26:
The makeBrowsePath syntax is decribed here : https://github.com/node-opcua/node-opcua/blob/master/lib/address_space/make_relative_path.js#L19

erossignon, 2016-12-06:
@gbaquedano any progress ?

gbaquedano, 2016-12-12:
Hi Etienne,
had no success with this. I cannot expose a server right now, but KepServerEX6 can be downloaded and used with its built in simulation driver to do this tests.

Thank you for your time.

erossignon, 2016-12-13:
I have install KepwareEx6 and confirm the issue. 
I have created a [gist that demonstrate that KepWareEX6](https://gist.github.com/erossignon/e5fa4b4133d16be6669411a790f20c61.js) cannot translate browsepath containing more than 1 element. The correctness of the browsePath have been cross-checked with UAExpert.

This is a conformance issue  and it shall be reported to Kepware.

closed 2018-01-25 by erossignon