node-opcua/node-opcua #276: npm test not work on latest version

wcc21320, 2016-11-22:
Hi, I just get code from git and run "npm test" and got the result:
1. missing files : 
"schemas/0617cedc0f7d3da5109a145acef074245fd44ca8"
"schemas/39394884f696ff0bf66bacc9a8032cc074e0158e"
'_generated_/_auto_generated_ServerStatus'
'_generated_/_auto_generated_ServerDiagnosticsSummary'
'_generated_/_auto_generated_ModelChangeStructure'
'schemas/39394884f696ff0bf66bacc9a8032cc074e0158e/TimeZone'
2. nodejs version limit on 0.12

Here's somthing detail:

For 1:
` Error: Cannot find module 'schemas/39394884f696ff0bf66bacc9a8032cc074e0158e/ServerState_enum'`

For 2:
I try to copy missing (no generated automatically) files from older version, and there's a nodejs version issue:

```
 mocha test --color -R spec --recursive --timeout 200000 --bail
vInt64_1000 Variant(Scalar<Int64>, value: 0,1000)
/home/wcc/store2/projects/aiot/gateway/opcua/node_modules/node-opcua/node_modules/node-opcua-crypto/lib/crypto_utils.js:53
        parts.push(Buffer.from(base64str, "base64"));
                          ^

TypeError: base64 is not a function

```
I use nodejs 4.4.7 in elder version is fine, but this version seems have to use v0.12.

And finally, still missing a file during npm test: 
`Error: Cannot find module 'schemas/39394884f696ff0bf66bacc9a8032cc074e0158e/TimeZone'`

FYR~

closed 2016-11-22 by erossignon

wcc21320, 2016-11-23:
Hi, Would you please tell us how to generate those missing files? Since npm test will failed without them.

erossignon, 2016-11-23:
```npm install```  should help. It is a good idea to always run  ```npm install``` after ```git pull```.

wcc21320, 2016-11-23:
Thanks a lot~
I found the reason: I run npm install by nodejs v4.4.7, seems no error but those files not be generated!
Now I switch to v0.12.13 and run npm install again then got them there.