node-opcua/node-opcua #277: error crash as a server register to OPCUADiscoveryServer

wcc21320, 2016-11-23:
Hi, I got an error crash as a server register to OPCUADiscoveryServer, here's the dump stack :
```
.../node_modules/node-opcua/node_modules/node-opcua-crypto/lib/crypto_utils.js:306
        return Buffer.alloc(1);
                      ^
TypeError: undefined is not a function
    at privateDecrypt_native (.../node_modules/node-opcua/node_modules/node-opcua-crypto/lib/crypto_utils.js:306:23)
    at Object.privateDecrypt_long (.../node_modules/node-opcua/node_modules/node-opcua-crypto/lib/crypto_utils.js:358:29)
    at Object.RSAPKCS1V15_Decrypt [as asymmetricDecrypt] (.../node_modules/node-opcua/lib/misc/security_policy.js:108:25)
    at MessageBuilder._decrypt_OPN (.../node_modules/node-opcua/lib/misc/message_builder.js:156:50)
    at MessageBuilder._decrypt (.../node_modules/node-opcua/lib/misc/message_builder.js:322:21)
    at MessageBuilder._read_headers (.../node_modules/node-opcua/lib/misc/message_builder.js:367:19)
    at MessageBuilderBase._append (.../node_modules/node-opcua/lib/misc/message_builder_base.js:114:15)
    at MessageBuilderBase._feed_messageChunk (.../node_modules/node-opcua/lib/misc/message_builder_base.js:173:14)
    at null.<anonymous> (.../node_modules/node-opcua/lib/misc/message_builder_base.js:49:14)
    at emit (events.js:107:17)
```

And I run the code under nodejs v0.12.13.

erossignon, 2016-11-23:
once again ```npm install``` could help.
Please note that node 0.12 will no longer be supported in the future, I recommand that you update to node 6.x.

wcc21320, 2016-11-23:
Thank you, I will upgrade it.

closed 2016-11-23 by wcc21320