node-opcua/node-opcua #278: Generate OPC AC Events

carlos2000iconics, 2016-11-24:
Hello

Do you have an example at generating AC Events from the SDK?

regards

Carl

erossignon, 2016-12-06:
Yes, we now have A&C. This is still work in progress.
Unit test are a good place to find example.
What sort of A&C are you trying to implement ?

carlos2000iconics, 2016-12-06:
Hello

I am try to generate events based on conditions of DA points, which need to
be acked knowledge.

Can you point me in the direction of the Unit tests please.

regards

Carl

erossignon, 2016-12-12:
You will find an example of implementing [an AcknowledgeableConditionType here](https://github.com/node-opcua/node-opcua/blob/master/test/address_space/alarms_and_conditions/utest_acknowledgeable_condition.js)

closed 2017-01-04 by erossignon