node-opcua/node-opcua #279: Read  OPC Items from array

sanlop73, 2016-12-02:
Good morning:
I am trying to execute the attached code and i am getting the following error:
[Error: Callback was already called.]
If i remove the instruction callback(err); in Step4 , then i am able to retrive the values from the entire array but the program does not exit, i need to press Crtl-C.

Any suggestios or ideas?

Thanks.

var opcua = require("node-opcua");
var async = require("async");
var client = new opcua.OPCUAClient();
var endpointUrl = "opc.tcp://xxx.xxx.xxx.xxx:4096";
var the_session, the_subscription;
var triggersOPs= ["ns=1;s=[LT2]DB101,I0",
	   "ns=1;s=[LT2]DB101,DI90",
	   "ns=1;s=[LT2]DB101,DI30",
	   "ns=1;s=[LT2]DB101,I2",
           "ns=1;s=[LT2]DB101,DI34",
           "ns=1;s=[LT2]DB101,DI94"
           ];

async.series([
    // step 1 : connect to
    function(callback)  {
        client.connect(endpointUrl,function (err) {
            if(err) {
                console.log(" cannot connect to endpoint :" , endpointUrl );
            } else {
                console.log("connected !");
            }
            callback(err);
        });
    },

    // step 2 : createSession
    function(callback) {
        client.createSession( function(err,session) {
            if(!err) {
                the_session = session;
            }
            callback(err);
        });
    },

    
    // step 4' : read a variable with read
    function(callback) {
var err;    
triggersOPs.forEach(function(entry) {
       var max_age = 0;
       var nodes_to_read = [
          { nodeId: entry, attributeId: opcua.AttributeIds.Value }
       ];
       the_session.read(nodes_to_read, max_age, function(err,nodes_to_read,dataValues) {
           if (!err) {
               console.log(entry , dataValues[0].value.value);
           }
           callback(err);
       });
     });
    },
    

    // close session
    function(callback) {
        the_session.close(function(err){
            if(err) {
                console.log("session closed failed ?");
            }
            callback();
        });
    }

],
function(err) {
    if (err) {
        console.log(" failure ",err);
    } else {
        console.log("done!");
    }
    client.disconnect(function(){});

erossignon, 2016-12-06:
In step 4 you are making a call to the async function reads for each element in the ```triggerOps``` array. As a consequence, ```callback``` is call for each read.
I suggest that you aggregates all the items to read into a single read request.

```javascript
// step 4' : read a variable with read
function(callback) {
     var err;    
     var nodes_to_read = [  ];
       
      triggersOPs.forEach(function(entry) {
           nodes_to_read.push({ nodeId: entry, attributeId: opcua.AttributeIds.Value });
      });

       var max_age = 0;
       the_session.read(nodes_to_read, max_age, function(err,nodes_to_read,dataValues) {
           if (!err) {
               for(var i;i<dataValues.length;i++) {
                      console.log(entry , dataValues[i].toString());
                }
           }
           // now callback is called once
           callback(err);
     });
    },
```

erossignon, 2017-01-04:
@sanlop73, any feedback ? can we close this issue now ?

closed 2017-03-31 by erossignon

sanlop73, 2017-05-03:
Sorry for the delay, had problems with my email.
I have checked and it works.

Thank you

joselopezbinoovo, 2023-03-29:
no