node-opcua/node-opcua #282: openssl date format error on certificate creation

MarcelDe, 2016-12-13:
Using 
`npm install `
or 
`node ./postinstall.js`
on OSX

generates the following error
```
"openssl" ca  -config conf/caconfig.cnf  -startdate 20161211205022Z -enddate 20171211205022Z -batch -out "/Users/marcel/Documents/projecten/sandbox/node-opcua/certificates/client_cert_2048.pem" -in "/Users/marcel/Documents/projecten/sandbox/node-opcua/certificates/client_cert_2048.pem.csr"
Using configuration from conf/caconfig.cnf
start date is invalid, it should be YYMMDDHHMMSSZ
```

Seems that the year part must be 2 digits long (but i'm not a openssl expert)
I've "fixed" this issue by modifying the last line of the function toolbox.x509Date to

` return w(Y, 2) + w(M, 2) + w(D, 2) + w(h, 2) + w(m, 2) + w(s, 2) + "Z";`

erossignon, 2016-12-13:
duplicates #235 #254

closed 2016-12-13 by erossignon

erossignon, 2016-12-13:
see also https://github.com/node-opcua/node-opcua/wiki/installing-node-opcua-or-node-red-on-MacOS