node-opcua/node-opcua #287: Problem with historizing

zenonpelle, 2016-12-19:
I can't get "historizing" work. Is there any example how it works?

BLamy, 2016-12-27:
[image]

Looking at the readme it seems like historical access isn't implemented yet. However you can record data yourself by subscribing to the data and inputting it into your own database such as MySQL or Influx db.

erossignon, 2016-12-29:
@zenonpelle  Can you provide more info ?
 Are you willing to implement an historizing server ? or implement a client accessing a historizing server ?

zenonpelle, 2017-01-02:
I was interested to use the historizing functionality för trends and alarm. Hvae you any idea howe historizing should be implemented server side. I could be interested to implement a historizing server

erossignon, 2017-01-03:
Simple InMemory variable Historization is available.
Here is a [example](https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_server_with_a_historizing_variable.md)

Let me know if this help.

zenonpelle, 2017-01-04:
thank you! It helps a lot!

closed 2017-01-04 by zenonpelle