node-opcua/node-opcua #288: Client reconnection after break

coussej, 2016-12-19:
Hi, 

After reading some of the older issues here, I'm still trying to figure out what the idiomatic way is to cope with a connection break. In my first setup, I simply handled every error coming from a session.read() or from a subscription by completely disconnecting everything and starting over by creating a new connection. However, as also mentioned in #144 , the client should be able to detect a connection break and attempt to re-establish the connection and the session. I don't get this to work however.

My test (very simple): 

create a client like this:

```javascript
this.uaClient = new opcua.OPCUAClient({
      connectionStrategy: {
        maxRetry: 3,
        initialDelay: 1000,
        maxDelay: 10000},
      keepSessionAlive: true
    })
```
connect to online UAServer, read a value every second, then turn of my internet connection. If I turn it back on a few seconds later, it continues working as expected. However, as soon as the underlying socket is closed (`this.uaClient._secureChannel._transport._socket.on("close", ...)`), the client does nog automatically reconnect. I have subscribed to all of these events:

```javascript
this.uaClient
      .on('start_reconnection', () => { console.log('start_reconnection') })
      .on('connection_reestablished', () => { console.log('connection_reestablished') })
      .on('close', () => { console.log('close') })
      .on('backoff', () => { console.log('backoff') })
```

However, none of those is triggered. Also, **uaclient. isReconnecting** is always false. It's not really clean to subscribe to these local events as @pelis already mentioned in #144. Is there anything I can do to have the client automatically reconnect after a socket break?

PiotrDeyna, 2016-12-20:
I wish to reconnect client after server restart. It doesn't work:

On client side:
start_reconnection not working so aborting
  connection failed for the 0  time ... We will retry in  1000  ms
  connection failed for the 1  time ... We will retry in  2000  ms
  connection failed for the 2  time ... We will retry in  4000  ms
connection_reestablished 
err =  [Error:  ServiceResult is BadSessionNotActivated (0x80270000)]
warning : disconnection : closing pending sessions
close and abort

on server side:
 Bad Session in  _on_ActivateSessionRequest <key>

erossignon, 2016-12-28:
@coussej  how do you "turn of the internet connection" ? 
   * unplug the cable ?
   * disable network card driver ?
   * shutdown the router ?

erossignon, 2016-12-28:
OK , I can reproduce it.
It seems that the reconnexion process is not completed and is lacking session reconstruction when tcp disconnection time exceed session lifetime.

erossignon, 2017-01-03:
fixed in v0.0.61

coussej, 2017-01-04:
I just tested this but reconnection is not happening. I still get the socket close event, but none of the others (start_reconnection,...) is triggered. Do I have to trigger this somehow?

PiotrDeyna, 2017-01-10:
I've installed v0.0.61 and it's working,
Thanks!

KENZoft, 2017-01-10:
If I shutdown the OPC-UA server (CoDeSys 3.5) and restart it then I get this error message:

Reconnect...
  connection failed for the 0  time ... We will retry in  1077  ms
  connection failed for the 1  time ... We will retry in  2083  ms
  connection failed for the 2  time ... We will retry in  4163  ms
  connection failed for the 3  time ... We will retry in  8286  ms
err =  { [Error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2017-01-10T15:10:40.594Z
   requestHandle               /* IntegerId                        */: 2               0x2
   serviceResult               /* StatusCode                       */: BadSessionIdInvalid (0x80250000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};]
  response:
   ServiceFault {
     responseHeader:
      ResponseHeader {
        timeStamp: Tue Jan 10 2017 16:10:40 GMT+0100 (Västeuropa, normaltid),
        requestHandle: 2,
        serviceResult: [Object],
        serviceDiagnostics: [Object],
        stringTable: [Object],
        additionalHeader: null } } }
warning : disconnection : closing pending sessions

How can I solve this? Do I need to make a completely new client to reconnect if I restart my OPC-UA server?

PiotrDeyna, 2017-01-10:
My code on the client side is:
...
var options = {
    connectionStrategy: {
        maxRetry: 100000
    }
  };
 ...
var opcuaClient = new opcua.OPCUAClient(options);
...

async.series([
        // step 1 : connect to
        function (callback) {
            console.log("connect to " + endpointUrl + " ... ");
            opcuaClient.connect(endpointUrl, function (err) {
                if (err) {
                    console.log(" cannot connect to endpoint :", endpointUrl);
                } else {
                    console.log("connected !");
                }
                callback(err);
            });
            opcuaClient.on("timed_out_request ", function () {
                console.log("timed_out_request ");
            });
            opcuaClient.on("start_reconnection", function () {
                console.log("start_reconnection not working so aborting");
            });
            opcuaClient.on("connection_reestablished", function () {
                console.log("connection_reestablished ");
            });
            opcuaClient.on("close", function () {
                console.log("close and abort");
                opcuaServerUp = false;
            });
            opcuaClient.on("backoff", function (nb, delay) {
                console.log("  connection failed for the", nb,
                        " time ... We will retry in ", delay, " ms");
                opcuaServerUp = false;
            });
        },

       // step 2 : createSession
        function (callback) {
            console.log("create Session !");
            opcuaClient.createSession(function (err, session) {
                if (!err) {
                    theOpcuaSession = session;
                }
                callback(err);
            });
        },

...

On the server side:

var server = new opcua.OPCUAServer({
...

function post_initialize() {
...
    function construct_my_address_space(server) {

        var addressSpace = server.engine.addressSpace;
...
    }

    construct_my_address_space(server);

    server.start(function () {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        console.log("port ", server.endpoints[0].port);

        var endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
        console.log(" the primary server endpoint url is ", endpointUrl);

    });
}

server.initialize(post_initialize);

Hope it helps ...

KENZoft, 2017-01-11:
Ok, thanks!

jugglingcats, 2017-06-14:
Should session (including monitored items etc) be automatically re-established after a network break or server restart? I am using Codesys server and see the following sequence:

```
 ... start_reconnection
start_reconnection not working so aborting
  connection failed for the 0  time ... We will retry in  1015  ms
  connection failed for the 1  time ... We will retry in  2192  ms
connection_reestablished
warning : disconnection : closing pending sessions
close and abort
```

I am triggering this by simply stopping and starting the Codesys PLC (not dropping network connection).

At this point connection is dead I think: items are no longer monitored.

Just wondering if the client is able to re-establish connection AND re-create subscription(s) and monitored items. Or we need to handle this ourselves. Tx

Razzeee, 2017-11-29:
@jugglingcats were you able to solve this problem I'm also getting `warning : disconnection : closing pending sessions`

jugglingcats, 2017-11-29:
No I wasn't but I didn't pursue it

nemethz5, 2018-02-01:
Is there a solution?

closed 2018-09-23 by erossignon

Zied91, 2020-03-31:
Hi @erossignon  

I am getting the below error when I restart matrikon OPC UA server service and re-connection fails However it reconnects correctly when I stop the server service then wait around 20/30 seconds and then start it again (btw I am using version 2.4.4)

Please see below the error log
-------------------------------------------------------------------------------------------------------
Connection lost
reconnection
still trying to connect: retry = 0, next attempt in 0.01 seconds
close
Disconnected following reconnection failure  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader      */: {
   timestamp                   /* DateTime            */: 2020-03-31T14:41:59.162Z
   requestHandle               /* UInt32              */: 8               0x8
   serviceResult               /* StatusCode          */: BadServerHalted (0x800e0000)
   serviceDiagnostics          /* DiagnosticInfo      */: { /*DiagnosticInfo*/
 namespaceUri                  /* Int32               */: null
 symbolicId                    /* Int32               */: -1
 locale                        /* Int32               */: -1
 localizedText                 /* Int32               */: -1
 additionalInfo                /* String              */: null
 innerStatusCode               /* StatusCode          */: Good (0x00000)
 innerDiagnosticInfo           /* DiagnosticInfo      */: null
};
   stringTable                 /* UAString         [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject     */: null
 }
}; request = { /*CreateSessionRequest*/
 requestHeader                 /* RequestHeader       */: {
   authenticationToken         /* NodeId              */: ns=0;i=0
   timestamp                   /* DateTime            */: 2020-03-31T14:41:54.024Z
   requestHandle               /* UInt32              */: 8               0x8
   returnDiagnostics           /* UInt32              */: 0               0x0
   auditEntryId                /* UAString            */:
   timeoutHint                 /* UInt32              */: 60000               0xea60
   additionalHeader            /* ExtensionObject     */: null
 }
 clientDescription             /* ApplicationDescri   */: {
   applicationUri              /* UAString            */: urn:Edge:OPCUA
   productUri                  /* UAString            */: NodeOPCUA-Client
   applicationName             /* LocalizedText       */: locale=null text=NodeOPCUA-Client
   applicationType             /* ApplicationType     */: 1 ( 1)
   gatewayServerUri            /* UAString            */:
   discoveryProfileUri         /* UAString            */:
   discoveryUrls               /* UAString         [] */: [ /* empty*/ ]
 }
 serverUri                     /* UAString            */: urn:EC2AMAZ-G0MV2KA:MatrikonOPC UA Wrapper for COM OPC Servers
 endpointUrl                   /* UAString            */: opc.tcp://1.14.114.254:21381/MatrikonOpcUaWrapper
 sessionName                   /* UAString            */: Session2
 clientNonce                   /* ByteString          */
BUFFER{00000000: 61 7b 6e 46 ca c1 57 dc 4f 9e 6f 18 0a 7d c4 ce 81 ec e1 06 28 16 8f 7b 18 7e 91 bc a7 f3 51 72    a{nFJAW\O.o..}DN.la.(..{.~.<'sQr
}
 clientCertificate             /* ByteString          */
BUFFER{00000000: 30 82 04 44 30 82 03 2c a0 03 02 01 02 02 09 00 b3 d2 d1 43 43 92 e1 99 30 0d 06 09 2a 86 48 86    0..D0..,........3RQCC.a.0...*.H.
00000020: f7 0d 01 01 0b 05 00 30 5f 31 0b 30 09 06 03 55 04 06 13 02 4e 4c 31 0b 30 09 06 03 55 04 08 0c    w......0_1.0...U....NL1.0...U...
00000040: 02 5a 48 31 0c 30 0a 06 03 55 04 07 0c 03 52 54 4d 31 13 30 11 06 03 55 04 0a 0c 0a 52 6f 79 61    .ZH1.0...U....RTM1.0...U....Roya
00000060: 6c 56 6f 70 61 6b 31 11 30 0f 06 03 55 04 0b 0c 08 47 6c 6f 62 61 6c 49 74 31 0d 30 0b 06 03 55    lVopak1.0...U....GlobalIt1.0...U
00000080: 04 03 0c 04 45 64 67 65 30 1e 17 0d 32 30 30 33 31 32 30 39 30 39 32 37 5a 17 0d 33 30 30 33 31    ....Edge0...200312090927Z..30031
000000a0: 30 30 39 30 39 32 37 5a 30 5f 31 0b 30 09 06 03 55 04 06 13 02 4e 4c 31 0b 30 09 06 03 
.........................................
 .... ( 1096)}
 requestedSessionTimeout       /* Double              */: 60000
 maxResponseMessageSize        /* UInt32              */: 800000               0xc3500
};

erossignon, 2020-04-19:
@Zied91 I would need to have access to the Matrikon server for investigation, can you contact me at contact@sterfive.com ?

xmariopereira, 2020-06-11:
Hi @erossignon,

I work with Zied, sorry for not giving feedback regarding this situation. But we are currently checking in the matrikon side :) We will contact you if necessary. Thank you for your support!