node-opcua/node-opcua #289: How to increase session timeout oder to change the default timeout

josiasyoumbi, 2017-01-03:
Hello everybody,

If i run my client, everything is okay, but after a few minutes nothing more, but I do not know if something has to do with session, i have this message: Timeout .... waiting for response for  CloseSecureChannelRequest. How to increase session timeout oder to change the default timeout

Thanks for their help
Best regards
Josias

erossignon, 2017-01-03:
Let's do not jump to quick into the conclusion that timeout need to be increased.

Does you client communicate regularly with the server  ?

The client shall have a least one transaction during the session timeout  timespan to prevent timeout.

This could be achieve by creating an empty subscription. Subscriptions have a keepalive mechanism that will cause at least one transaction every keepaliveCount*publishingInterval ms.

Alternatively , if your server doesn't provide subscriptions (nano server for instance) you could use the 
```keepSessionAlive: true``` options. When ```keepSessionAlive``` is set to ```true```, ```OPCUAClient``` regularly interogate the ```currentTime``` variable with a ```readv command to prevent the session timeout.

```javascript
var client = new opcua.OPCUAClient({
   keepSessionAlive:  true
  //
});
```
This being said,
the session time out is negotiated between the client and the server during the session creation by specifying the ```requestedSessionTimeout``` options during ```OPCUAClient``` creation.

```javascript
var client = new opcua.OPCUAClient({
   keepSessionAlive:  true,
   requestedSessionTimeout: 1200000 //milliseconds (default was 60x1000 milliseconds )
  //
});
```

josiasyoumbi, 2017-01-04:
Waou waou thank you very much  for an answer, i used the **keepSessionAlive: true options**. it works very well now. that's brilliant

closed 2017-01-04 by erossignon