node-opcua/node-opcua #290: version 0.61

zenonpelle, 2017-01-04:
When installing the 0.61 version I get the the foloowing when I am starting my application:

 node sampleserver.js
module.js:340
    throw err;
    ^

Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:289:25)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (C:\Users\Zenon\OneDrive\utveckling\opcuaserver\node_modules\node-opcua\lib\server\server_engine.js:69:20)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)

erossignon, 2017-01-04:
We probably need more information here to understand what's going on:
* What version of node are you using ?
* What OS ? ( Linux, Mac, other ?)
* Did you install node-opcua@0.0.61 from npm or from git ?
   * npm install node-opcua ?
   * git clone  ?
* If you used git clone, is it on  a fresh installed or on top of a old version ?

zenonpelle, 2017-01-04:
I use windows 10 as operating system.
version 4.2.4 of node.js
I installed the new verison with npm install node-opcua.

remove node-opcua and make an install of the latest version ant then the error appears. Install text below:
 npm install

> node-opcua@0.0.61 postinstall C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua
> node postinstall.js

Executing  node bin/generate_opcua_classes.js --clear
done ... (1)
Executing  node bin/crypto_create_CA.js demo --silent

        Create Demo certificates

reading configuration
=====================

 Creating default g_config file  C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\certificates\config.js
 configuration =
   subject                        : /C=FR/ST=IDF/L=Paris/O=NodeOPCUA/CN=NodeOPCUA-TEST
   validity                       : 5475
   keySize                        : 2048
   certificateDir                 : C:/Users/Zenon/OneDrive/utveckling/opcuahtml/node-opcua-htmlpanel-master/node_modules/node-opcua/certificates
   CAFolder                       : C:/Users/Zenon/OneDrive/utveckling/opcuahtml/node-opcua-htmlpanel-master/node_modules/node-opcua/certificates/CA
   PKIFolder                      : C:/Users/Zenon/OneDrive/utveckling/opcuahtml/node-opcua-htmlpanel-master/node_modules/node-opcua/certificates/PKI
   altNames                       :
   dns                            : PELLESPLATTA
   ip                             :
creating openssl_folder C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl
checking presence of  C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\openssl.exe
 cannot find file C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\openssl.exe
openssl seems to be missing and need to be installed
downloading http://indy.fulgan.com/SSL/openssl-1.0.2j-x64_86-win64.zip
deflating  C:\cygwin64\tmp\openssl-1.0.2j-x64_86-win64.zip
 unzipping : C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\HashInfo.txt
 unzipping : C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\libeay32.dll
 unzipping : C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\OpenSSL License.txt
 unzipping : C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\openssl.exe
 unzipping : C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\ReadMe.txt
 unzipping : C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\ssleay32.dll
unzip done
verifying  true OK  C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\openssl.exe
done
checking presence of  C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\openssl.exe
        stdout OpenSSL 1.0.2j  26 Sep 2016
 Version =  OpenSSL 1.0.2j  26 Sep 2016
openssl_path C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\node-opcua-pki\bin\openssl\openssl.exe
OpenSSL version :  OpenSSL 1.0.2j  26 Sep 2016
     hostname =  PELLESPLATTA
 urn =  urn:PELLESPLATTA:NodeOPCUA-Client
 urn =  urn:PELLESPLATTA:NodeOPCUA-Client
 urn =  urn:PELLESPLATTA:NodeOPCUA-Server
 urn =  urn:PELLESPLATTA:NodeOPCUA-Server
 urn =  urn:PELLESPLATTA:NodeOPCUA-DiscoveryServer
 urn =  urn:PELLESPLATTA:NodeOPCUA-DiscoveryServer
done ... (0)
node-opcua@0.0.61 node_modules\node-opcua
├── easy-table@1.0.0
├── byline@5.0.0
├── set-prototype-of@1.0.0
├── sprintf@0.1.5
├── bomstrip@0.1.4
├── dequeue@1.0.5
├── underscore@1.8.3
├── progress@1.1.8
├── q@1.4.1
├── humanize@0.0.9
├── treeify@1.0.1
[... 14 lines cut ...]
├── node-opcua-pki@0.0.11 (walk@2.3.9, cli-table@0.3.1, yauzl@2.7.0, del@2.2.2)
├── deassertify@0.1.2 (minimist@1.2.0, through2@2.0.3, falafel@2.0.0)
├── node-opcua-crypto@0.0.7 (ssh-key-to-pem@0.11.1)
├── yargs@6.6.0 (get-caller-file@1.0.2, camelcase@3.0.0, decamelize@1.2.0, y18n@3.2.1, which-module@1.0.0, set-blocking@2.0.0, yargs-parser@4.2.1, require-directory@2.1.1, require-main-filename@1.0.1, os-locale@1.4.0, cliui@3.2.0, string-width@1.0.2, read-pkg-up@1.0.1)
├── moment@2.17.1
└── async@2.1.4 (lodash@4.17.4)

Zenon@PellesPlatta /cygdrive/c/Users/Zenon/OneDrive/utveckling/opcuahtml/node-opcua-htmlpanel-master
$ node app.js
module.js:340
    throw err;
    ^

Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:289:25)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\lib\server\server_engine.js:69:20)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)

erossignon, 2017-01-04:
can you try  this:
```
npm uninstall node-opcua
npm install node-opcua
```
If not conclusive, can you then try:
```
node node_modules\node-opcua\bin\generate_opcua_classes.js --verbose --clear
```

zenonpelle, 2017-01-04:
Hi!
It did not help
Result from generate script:
$ node node_modules/node-opcua/bin/generate_opcua_classes.js --verbose --clear
removing files in  C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AcknowledgeMessage.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_ActivateSessionRequest.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_ActivateSessionResponse.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AddNodesItem.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AddNodesRequest.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AddNodesResponse.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AddNodesResult.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AddReferencesItem.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AddReferencesRequest.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AddReferencesResponse.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AggregateConfiguration.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AnonymousIdentityToken.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_ApplicationDescription.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_Argument.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AsymmetricAlgorithmSecurityHeader.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AttributeOperand.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_AxisInformation.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowseDescription.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowseNextRequest.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowseNextResponse.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowsePath.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowsePathResult.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowsePathTarget.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowseRequest.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowseResponse.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BrowseResult.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_BuildInfo.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_CallMethodRequest.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_CallMethodResult.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_CallRequest.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_CallResponse.js
.... deleting   C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\_generated_/_auto_generated_CancelRequest.js
[... cut ...]
    at SaxParser.parseString (C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\ersatz-node-expat\node_modules\node-xml\lib\node-xml.js:829:10)
    at Parser.write (C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\node_modules\ersatz-node-expat\lib.js:75:15)
    at C:\Users\Zenon\OneDrive\utveckling\opcuahtml\node-opcua-htmlpanel-master\node_modules\node-opcua\lib\xml2json\lib.js:287:24
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:380:3)

Best regards
Per

zenonpelle, 2017-01-04:
Hi!
I upgaded node.js and then it works fine!

Best regards
Per

closed 2017-01-04 by erossignon

okansari91, 2017-02-13:
Hi @zenonpelle, i am also facing the same issue and maybe you can help me out as you are done resolving. My current version of node is 4.4.0. Which version did you upgraded to? and with what command?

Moreover, whenever I perform "npm install node-opcua", i got these errors regarding OpenSSL. Does this need to be catered as well?

`OpenSSL version :  OpenSSL 1.0.2j  26 Sep 2016
ERROR  Command failed: C:\WINDOWS\system32\cmd.exe /s /c ""D:\NodeJS OPC UA\SimpleServerClient\myserver\node_modules\node-opcua-pki\bin\openssl\openssl.exe" ca -gencrl  -config conf/caconfig.cnf  -out crl/revocation_list.crl"
Using configuration from conf/caconfig.cnf
Error opening CA private key D:/NodeJS OPC UA/SimpleServerClient/myserver/node_modules/node-opcua/certificates/CA            # the main CA folder/private/cakey.pem
5872:error:02001003:system library:fopen:No such process:.\crypto\bio\bss_file.c:402:fopen('D:/NodeJS OPC UA/SimpleServerClient/myserver/node_modules/node-opcua/certificates/CA            # the main CA folder/private/cakey.pem','rb')
5872:error:20074002:BIO routines:FILE_CTRL:system lib:.\crypto\bio\bss_file.c:404:
unable to load CA private key`

I am using 
OS: Windows 10
Node: v4.4.0
Npm: 3.8.2

Thanking You!

DynamicRemo, 2017-02-13:
Hi @zenonpelle, i am also facing the same issue and maybe you can help me out as you are done resolving. My current version of node is 4.4.0. Which version did you upgraded to? and with what command?

Moreover, whenever I perform "npm install node-opcua", i got these errors regarding OpenSSL. Does this need to be catered as well?

`OpenSSL version : OpenSSL 1.0.2j 26 Sep 2016 ERROR Command failed: C:\WINDOWS\system32\cmd.exe /s /c ""D:\NodeJS OPC UA\SimpleServerClient\myserver\node_modules\node-opcua-pki\bin\openssl\openssl.exe" ca -gencrl -config conf/caconfig.cnf -out crl/revocation_list.crl" Using configuration from conf/caconfig.cnf Error opening CA private key D:/NodeJS OPC UA/SimpleServerClient/myserver/node_modules/node-opcua/certificates/CA # the main CA folder/private/cakey.pem 5872:error:02001003:system library:fopen:No such process:.\crypto\bio\bss_file.c:402:fopen('D:/NodeJS OPC UA/SimpleServerClient/myserver/node_modules/node-opcua/certificates/CA # the main CA folder/private/cakey.pem','rb') 5872:error:20074002:BIO routines:FILE_CTRL:system lib:.\crypto\bio\bss_file.c:404: unable to load CA private key`

I am using
OS: Windows 10
Node: v4.4.0
Npm: 3.8.2

Thanking You!

erossignon, 2017-02-14:
This issue is related to node version 4  which will no longer be supported.

please upgrade to **node version 6 or 7** 

Here is the error appearing when using node 4.x.x:
```
                    var byteString = Buffer.from(base64text,"base64");
                                            ^
TypeError: base64 is not a function
    at Function.from (native)
    at Function.from (native)
    at ReaderState.state_Variant.parser.ByteString.finish [as _finish] 
```

DynamicRemo, 2017-02-14:
Thank you @erossignon, upgrading the Node to v6.9.5 resolves the Generation of Classes but still the certificates error are yet to be resolved.

But this command `node node_modules\node-opcua\bin\crypto_create_CA.js demo` for the Generation of Certificates still generates OpenSSL Error as i have mentioned in my previous comment. I think i need to resolve this via explicitly installing OpenSSL, can find any particular solution yet.