node-opcua/node-opcua #291: 0.0.61 Can't promisify anymore.

coussej, 2017-01-04:
I can't promisify the node-opcua library anymore since 0.0.61. I used to do this:

```javascript
let promise = require('bluebird')
let opcua = require('node-opcua')
promise.promisifyAll(opcua)
```

but now I get an error: `Cannot promisify an API that has normal methods with 'Async'-suffix`. It seems to have to do with a method called `readValueAsync` in `lib/address_space/ua_variable.js`. However, after comparing with 0.0.60 this method was already there.

It's probably related to commit 3765aa5, where `UAVariable` is now exported on package level, which causes the error.

The solution is either to add a filter to the `promisifyAll` method to exclude this method, or to use a custom suffix alltogether.

I know the node-opcua library doesn't make any commitment towards compatibility with promisification, but as this is a fairly common practice I thought I'd give you a heads up. Writing everything using async/await in node 7 (--harmony) really improves readability!

erossignon, 2017-01-04:
I  think we should really move toward ES6 for readility of code and use of Promise as well. This would mean dropping support for old version of nodejs. Please vote with a 👍 if you don't care switching on node 6 or above.

erossignon, 2017-01-04:
@coussej  shall we introduce a api break and rename the ```readValueAsync```  ? or is it easy to work around this issue with ```bluebird``` ?

erossignon, 2017-01-04:
@coussej , what about adding a new unit test that verifies and makes surethat promisification of node-opcua is alway up and running ? Could you contribute ?

coussej, 2017-01-04:
@erossignon The solution is easy:

```javascript
let promise = require('bluebird')
let opcua = require('node-opcua')
promise.promisifyAll(opcua, {
  filter: function (name) {
    return name !== 'readValueAsync'
  }
})
```
Of course, this means the readValueAsync is not promisified.

Another possibility is:
```javascript
let promise = require('bluebird')
let opcua = require('node-opcua')
promise.promisifyAll(opcua, { suffix: 'Asyncified' })
```
... and then  for example the `connect` method would be `connectAsyncified` instead of `connectAsync`.

coussej, 2017-01-04:
@erossignon if you decide to change the name of the method then I can certainly write a test for that!

erossignon, 2017-01-04:
simple workaround 
```javascript
var promise = require("bluebird");
let opcua = require('node-opcua');
opcua.UAVariable = null;
promise.promisifyAll(opcua);
```

coussej, 2017-01-04:
@erossignon For your question on moving to ES6, wouldn't it be better to create a dedicated issue?

erossignon, 2017-01-04:
yep, let me do this

closed 2017-04-01 by erossignon