node-opcua/node-opcua #292: Moving toward ES6

erossignon, 2017-01-04:
I think we should really move toward ES6 for readility of code and use of Promise as well and other nice stuff (import, modules ) 
This would mean dropping support for old version of nodejs. 
Please vote with a 👍 if you don't care switching on node 6 or above or 👎 if you would rather not now.

zenonpelle, 2017-01-05:
Hi!I
My vote is 👍

/Per

BLamy, 2017-01-08:
I would love to see this. Since the architecture is very OOP it may be worth looking into typescript shouldn't be that much more effort because most data types are already defined in the docs. 

If going with plain ES6 I would highly suggest using AirBNBs eslint rules. I've had very good success with upgrading ES5 code to ES6 using those rules.

gilesbradshaw, 2017-01-15:
I think it's a great idea to switch - however do you have to drop old versions of node?  Pretty sure you'd be able to compile your ES6 to ES5 with babel..

typescript is a great idea too..

gilesbradshaw, 2017-01-23:
I've got quite far doing an ES6 conversion - it would be good if you had a look  https://github.com/ua-ql/node-opcua

It doesn't build and install automatically quite properly yet but it will soon.  

I've web packed it to one bundle so pretty sure it will run on previous versions of node + wold be possible to use  ES2016+ and typescript etc.

There are a few things that needed changing - mainly do do with dynamic requires (you cant do things like require(avariable) as it sucks in everything as a possible match (ie all the test code) also this means you can't build and then require stuff 'on the fly' (you probably *could* using hot reloading) so I've split 'createObject) from registerObject and in places you have circular dependencies and things like exporting created schemas from tests which then use those schemas.

However I have got the code through the tests (modified to cope with webpack/import restrictions and also run a server and a client with it...

I am gong to carry on working on it - eg use default exports where appropriate and rename the modules the exported object etc. to however you're free to use what I have done or take a look if you like.  It definitely shows that it's practical.

gilesbradshaw, 2017-01-23:
ok so if you do...

````
git clone https://github.com/ua-ql/node-opcua-sampleserver.git
cd node-opcua-sampleserver
npm i
npm start
````

and...

````
git clone https://github.com/ua-ql/node-opcua-sample.git
cd node-opcua-sample
npm i
npm start
````

then you have webpacked ES6 client and server talking to eachother

and if you stop the server on port 1234 and do

````
git clone https://github.com/ua-ql/node-opcua.git
cd node-opcua
npm i
npm test
````

then it runs the full test suite on the ES6 code (not webpacked)

erossignon, 2017-01-23:
Great Work. Give me sometime I am quite busy right now. I shall review the code and prepare a merge.

It looks like Istanbul (codecoverage)  doesn't like ES6 ( from travis log)
istanbul cover -x "tmp/**" ./node_modules/mocha/bin/_mocha -- -R spec --recursive --timeout 100000 --bail test

gilesbradshaw, 2017-01-24:
Great.   I''ll try and loom at code coverage. There's a lot of changes.  Also suspect the documentation will now be broken - I've not looked at that.  I'm gong to keep on plugging away at it - such as using more default exports and naming files by the class names so it's easier to find your way around.

As @Blamy mentioned it might be a good idea to use typescript (or maybe flow?) too eventually.

gilesbradshaw, 2017-01-24:
I assume things like this https://github.com/node-opcua/node-opcua/blob/master/lib/address_space/alarms_and_conditions/trip_alarm.js are unfinished..

erossignon, 2017-01-24:
Giles your contribution is invaluable as well

I have forked you repo, and run ```npm test```  on master branch and got some errrs => the one you reported. 
I have some spurious output , I guess this is some remaining debug stuff  that need cleaning.

```
creating { schema: 'QueryNextRequest',
  optional_folder: undefined,
[...]
reading data2 D:\projet\node-opcua-ES6\nodesets\Opc.Ua.NodeSet2.xml
file read { data: <Buffer ef bb bf 3c 3f 78 6d 6c 20 76 65 72 73 69 6f 6e 3d 22 31 2e 30 22 20 6
  err: null }
data isss <Buffer 3c 3f 78 6d 6c 20 76 65 72 73 69 6f 6e 3d 22 31 2e 30 22 20 65 6e 63 6f 64 69
[..]
```
- Typescript would require an other massive refactoring. let digest this one first.

- trip_alarm.js  is not really unfinished , as the specs says it is just a subtype of UAOffNormalAlarm use for categorisation. It may evolve in the future.

 we need to find a smooth way to merge and make the transition.

- I would have been inclined to introduce ES6 features following their implementation in of nodejs , so we could always run node-opcua without transpiling. After thinking about it, this would void the use of ```import ```  statement unfortunately.
- We have to be careful not to introduce a performance hit by using babel transpilation on every single file at runtime. I like to keep good performance on low spec computer such as Rapsberry, Beagle, Chips etc ... 
- we would have to test using babel-cli to create ES5 ready to use library for legacy application or the one running on old version of nodes.
What's about creating a "ES6" branch in the main node-opcua repo so we could collaborate together in bringing this version to the required level ?

gilesbradshaw, 2017-01-24:
Hi yes I out various console.logs in but will remove them every now and again.

With all the files webpacked in bundle.js there;s no runtime file loading (except the one) so I *think* it should be faster or as fast.  + *should* run on all versions of node.

Yes happy to use ES6 branch of your main repository.  Maybe if I get "to a point" with what I am doing then put it in there.  I might just clone your repository - overwirite it with my code and push it to ES6 in one fell swoop so you don't have hundreds of commits.  We could always look back to ua-ql if we needed to dive into history.

Yes you're right about typescript - best to do this refactor first.  I should be done with current refactor in maybe a week..  It might be a bit hard for you to trace everything through though but I think it's normally pretty obvious if things work.

erossignon, 2018-10-28:
The next branch is now using TypeScript ! work is 80% done ..

closed 2018-10-28 by erossignon