node-opcua/node-opcua #293: Connect to multiple servers

josiasyoumbi, 2017-01-05:
Hi 
How to connect to more than one server, eg read alarms from multiple servers from a node opc ua client. Is it possible with node ua?

thank you in advance

erossignon, 2017-01-06:
Yes .It should be possible to acheive this by instantiating a OPCUAClient object per  server.

josiasyoumbi, 2017-01-31:
I do not have much experience with javascript and I would like to connect to several Servers. But I have a problem, I do not know if I declare bad the_sessions.
`var opcua = require("node-opcua");
var async = require("async");

var express = require('express')
    , app = module.exports = express();
	
var evilscan = require('evilscan');   // um das Netzwerk zu scannen

var options = {
    target:'192.168.0.0/24',
    port:'16664',
	reverse:true ,
    status:'O', //T Timeout,R Refused, O Open,U Unreachable
    banner:true
};
//var indexRoutes = require('./routes/index');
var router = express.Router();

var path = require('path');

var opt = {
    keepSessionAlive:  true
	//
  };
  
var client= [];
var sessions=[];
for (i = 0; i < 5; i++) { 
     client[i] = new opcua.OPCUAClient(opt);

}

var the_sessions = [];
for (i = 0; i < 5; i++) { 
     the_sessions[i] = null;
}
var availableRCEInfoNodeID  = "ns=1;i=62546";
var rootObjectNodeID        = "ns=0;i=85";

var global = "";
var results = [];		
var scanner = new evilscan(options)
async.series([

    // step 1 : connect to
    function(callback)  {
		
        
		
		scanner.on('result',function(data) {
					results.push(data);			
		});

		scanner.on('error',function(err) {
				throw new Error(data.toString());
				console.log(" cannot connect to endpoint :" , options.target );
			    callback(err);
		});

		scanner.on('done',function() {
				// finished !
        if(results == ""){
			console.log("Kein RCE läuft");
			
		}else{
			global = results;
			console.log("Anzahl RCE in diesem Netzwerk"+global.length);
            for (i = 0; i < global.length; i++) { 
		    client[i].connect("opc.tcp://"+results[0].ip+":"+results[0].port,function (err) {

				if(err) {
					console.log(" cannot connect to endpoint :" , results[0].ip+":"+results[0].port );
				} else {
					console.log("connected !");
				}
				callback(err);
            });
			}
		}
		});
		scanner.run();
    },
    // step 2 : createSession
    function(callback) {
		for (i = 0; i < global.length; i++) { 
			client[i].createSession( function(err,session) {
				if(!err) {
					the_sessions[i] = session;
					console.log("Session");
				}
				callback(err);
			});
		}
    },
    
    // -----------------------------------------
    // create subscription
    function(callback) {
			   
				// having to name the views as *.ejs
				app.engine('.html', require('ejs').__express);
				 
				// Set the folder where the pages are kept
				app.set('views', __dirname + '/views');
				
				app.use(express.static(__dirname + '/img'));
				 
				// This avoids having to provide the 
				// extension to res.render()
				app.set('view engine', 'html');
				
				
				
				// Serve the index page network Liste
				app.get('/', function(req, res){
					var methodsToCall = [];
                     methodsToCall.push({
                         objectId: rootObjectNodeID,
                         methodId: availableRCEInfoNodeID,
                     });
                		for (i = 0; i < global.length; i++) { 
							the_sessions[i].call(methodsToCall,function(err,results){
								string = JSON.parse(results[0].outputArguments[0].value);				
								console.log("done! %=",global);
		 
							});
						}
			
				});
						
				if (!module.parent) {
				  app.listen(8090);
				  console.log(' started, das Link http://localhost:8090/ aufrufen');
				}
    },

    // ------------------------------------------------
    // closing session
    
    function(callback) {
        console.log(" closing session");
		for (i = 0; i < global.length; i++) { 
        the_sessions[i].close(function(err){
            console.log(" session closed");
            callback();
        });
		}
    },

],

    function(err) {
        if (err) {
            console.log(" failure ",err);
        } else {
            console.log("done!")
        }
        client.disconnect(function(){});
    }) `

I have this error:
**TypeError: Cannot read property 'call' of null
    at D:\autodiscover_client\autodiscover.js:133:23**

erossignon, 2017-02-01:
In your async steps, the callback method is called more than once,
```javascript
function (callback) {
    for(i=0;i<n;i++) {
         calllback(); // wrong callbakc is called more than once
    }
}
```
Please study the [async](http://caolan.github.io/async/)

erossignon, 2017-02-01:
Closing as this question now derived to generic nodejs understanding.. please refer to related documentation.

closed 2017-02-01 by erossignon

gcperrin, 2017-03-21:
This seems like a design concern to me -- planning to take a look at the foundation stack, but I was under the impression that one client should be able to handle multiple sessions. Every other SDK I've worked with functions in this manner, assigning and managing session IDs for different endpoints. @erossignon any thoughts on the matter?

erossignon, 2017-03-21:
@gcperrin : this issue was related to a misunderstanding of javascript and the async design pattern. (the title is misleading, let me fix this)
please be reassure than node-opcua allows creation of multiple client connection to multiple server from a single applicationn, as many other SDKs.

gcperrin, 2017-03-21:
@erossignon Yes, apologies for the lack of clarity...you can connect to multiple servers...I have a working application that does achieve such with node-opcua. 

However, it still requires a **new client instance** for every connection, which is what seems misleading. My point about the other SDKs is around the fact that they create one client instance, that manages many session instances through things like the _sessionId._ returned from he server.

As I'm sure you know, much of the specification (things like Secure Channels) are based on communication between single client and a single server. Surely node-opcua handles this correctly through some internal mechanism, but to a user of the SDK the naming/nomenclature is confusing in this regard.