node-opcua/node-opcua #294: openssl error

AndreCatarinoGuerra, 2017-01-05:
C:\Users\André>npm install node-opcua

> node-opcua@0.0.61 postinstall C:\Users\André\node_modules\node-opcua
> node postinstall.js

Executing  node bin/generate_opcua_classes.js --clear
done
done ... (0)
Executing  node bin/crypto_create_CA.js demo --silent

        Create Demo certificates

reading configuration
=====================

 Creating default g_config file  C:\Users\André\node_modules\node-opcua\certificates\config.js
 configuration =
   subject                        : /C=FR/ST=IDF/L=Paris/O=NodeOPCUA/CN=NodeOPCUA-TEST
   validity                       : 5475
   keySize                        : 2048
   certificateDir                 : C:/Users/André/node_modules/node-opcua/certificates
   CAFolder                       : C:/Users/André/node_modules/node-opcua/certificates/CA
   PKIFolder                      : C:/Users/André/node_modules/node-opcua/certificates/PKI
   altNames                       :
   dns                            : DESKTOP-249EU16
   ip                             :
creating openssl_folder C:\Users\André\node_modules\node-opcua-pki\bin\openssl
checking presence of  C:\Users\André\node_modules\node-opcua-pki\bin\openssl\openssl.exe
 cannot find file C:\Users\André\node_modules\node-opcua-pki\bin\openssl\openssl.exe
openssl seems to be missing and need to be installed
downloading http://indy.fulgan.com/SSL/openssl-1.0.2j-i386-win32.zip

Finished writing to disk
deflating  C:\Users\André\AppData\Local\Temp\openssl-1.0.2j-i386-win32.zip
 unzipping : C:\Users\André\node_modules\node-opcua-pki\bin\openssl\HashInfo.txt
 unzipping : C:\Users\André\node_modules\node-opcua-pki\bin\openssl\libeay32.dll
 unzipping : C:\Users\André\node_modules\node-opcua-pki\bin\openssl\OpenSSL License.txt
 unzipping : C:\Users\André\node_modules\node-opcua-pki\bin\openssl\openssl.exe
 unzipping : C:\Users\André\node_modules\node-opcua-pki\bin\openssl\ReadMe.txt
 unzipping : C:\Users\André\node_modules\node-opcua-pki\bin\openssl\ssleay32.dll
unzip done
verifying  true OK  C:\Users\André\node_modules\node-opcua-pki\bin\openssl\openssl.exe
done
checking presence of  C:\Users\André\node_modules\node-opcua-pki\bin\openssl\openssl.exe
        stdout OpenSSL 1.0.2j  26 Sep 2016
 Version =  OpenSSL 1.0.2j  26 Sep 2016
openssl_path C:\Users\André\node_modules\node-opcua-pki\bin\openssl\openssl.exe
OpenSSL version :  OpenSSL 1.0.2j  26 Sep 2016
ERROR  Command failed: "C:\Users\André\node_modules\node-opcua-pki\bin\openssl\openssl.exe" ca -gencrl  -config conf/caconfig.cnf  -out crl/revocation_list.crl
reason: No such file or directory
Using configuration from conf/caconfig.cnf
unable to rename C:/Users/André/node_modules/node-opcua/certificates/CA/crlnumber.new to C:/Users/André/node_modules/node-opcua/certificates/CA/crlnumber

Command failed: "C:\Users\André\node_modules\node-opcua-pki\bin\openssl\openssl.exe" ca -gencrl  -config conf/caconfig.cnf  -out crl/revocation_list.crl
reason: No such file or directory
Using configuration from conf/caconfig.cnf
unable to rename C:/Users/André/node_modules/node-opcua/certificates/CA/crlnumber.new to C:/Users/André/node_modules/node-opcua/certificates/CA/crlnumber

done ... (0)

The folder exist, has several files and the executable opensll.exe...so I cannot figure it out.

erossignon, 2017-01-06:
Can you try inside a folder that doesn't contain accentuated character ?
( c:\User\André => c:\Projects\MyOpcuaProject)

AndreCatarinoGuerra, 2017-01-06:
Yes, it is the cause. What can be done to solve that?

erossignon, 2017-01-07:
The reason for this issue is that we use the port of openssl to windows to generate certificate. Unfortunately, Windows and Linux path name do not use the same encoding of special characters.
This could potentially be resolved  by using a tool like [iConv](http://gnuwin32.sourceforge.net/packages/libiconv.htm). But this is adding a extra layer of complexity.
I would keep the current behavior with the known limitation that node-opcua projet must be installed on pure ascii pathname. ... unless someone want to volunteer to contribute a fix.

closed 2018-01-28 by erossignon