node-opcua/node-opcua #297: Monitoring a method,

josiasyoumbi, 2017-01-12:
Hi 
in example on the internet, I do not see how to monitoring a method.

`        // install monitored item
        var monitoredItem = the_subscription.monitor({
            nodeId: opcua.resolveNodeId("ns=1;s=62541"),
            attributeId: opcua.AttributeIds.Value
        },
            {
                samplingInterval: 100,
                discardOldest: true,
                queueSize: 10
            },
            opcua.read_service.TimestampsToReturn.Both
        );
        console.log("-------------------------------------");

        monitoredItem.on("changed", function (dataValue) {
            console.log("methode = ", dataValue[0].outputArguments[0].value);
        });`
I try like this but it does not work, should I change the argument **attributeId**?

erossignon, 2017-01-25:
Method's  cannot be monitored. 
Monitoring is for  Variables data change notification or for Event notification.
What is the node ```ns=1;s=62541``` in your case ?

josiasyoumbi, 2017-01-31:
ns=1;s=62541 is Method

erossignon, 2017-02-01:
Method can only be called, see [example in unit test](https://github.com/node-opcua/node-opcua/blob/692ae94f24351909b152ac789e20b03fe2c4eb6a/test/end_to_end/u_test_e2e_issue_223_demonstrate_client_call_service.js#L86)

closed 2017-02-01 by erossignon