node-opcua/node-opcua #304: Any plans to convert from yuidoc to jsdoc?

gillerman, 2017-02-03:
I believe this would be the easiest way to support typescript functionality (at least using microsoft IDEs).

erossignon, 2017-02-05:
I have initiated the typescript definition file for node-opcua ( node-opuca.d.ts )
The following gist is a example of a typescript node-opcua client (with very limited functionality)
 https://gist.github.com/erossignon/282de0912e32a66f47a41da36a5a9acb
It should demonstrate that  we can get better intellisense support this way.

jugglingcats, 2017-06-13:
@erossignon I just created pull request #357 to cover monitoring of items in Typescript client. Hope this is appreciated.

jugglingcats, 2017-06-14:
Also #359 - 'on' method for OPCUAClientBase

bompi88, 2017-08-28:
I get the following errors when using `noImplicitAny: true`:

```bash
node_modules/node-opcua/node-opcua.d.ts(168,68): error TS7006: Parameter 'callback' implicitly has an 'any' type.
node_modules/node-opcua/node-opcua.d.ts(252,5): error TS7008: Member 'Good' implicitly has an 'any' type.
node_modules/node-opcua/node-opcua.d.ts(253,5): error TS7008: Member 'BadWaitingForInitialData' implicitly has an 'any' type.
```

erossignon, 2018-01-28:
Typescript is now supported, improvement and typescript coverage in a on going task.
let's close this issue.

closed 2018-01-28 by erossignon